usnistgov/ARIAC #173: still seg faults

db7296, 2022-05-26:
With new code, still get a seg fault when submitting a kitting order while an assembly order is also being filled.

First valid point will be reached in 0.559s.
[ INFO] [1653545644.448641538, 115.408000000]: [SUCCESS] [agv2: ks2->as1 triggered]
Segmentation fault
================================================================================REQUIRED process [gazebo-2] has died!
process has died [pid 4735, exit code 139, cmd /home/dbarry/ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode -r --record_period 0.01 --record_path=/home/dbarry/.ariac/log/gazebo /tmp/ariac/ariac.world __name:=gazebo __log:=/home/dbarry/.ros/log/b92a4e94-dcb9-11ec-9354-181deaec5b6c/gazebo-2.log].
log file: /home/dbarry/.ros/log/b92a4e94-dcb9-11ec-9354-181deaec5b6c/gazebo-2*.log
Initiating shutdown!
================================================================================

db7296, 2022-05-26:
This is with code changed according to the recent update guidelines for shutdown.

zeidk, 2022-05-26:
The performance.log file is broken and I cannot open it. Any way you can email it to me again?

db7296, 2022-05-26:
Full score with all 4 parts.  Nice fix!

zeidk, 2022-05-27:
Actually found the issue. I will do a new push as soon as I can.

db7296, 2022-05-31:
The new build eliminates these seg faults

closed 2022-05-31 by db7296