usnistgov/ARIAC #175: segmentation fault

liuhaoran1, 2022-05-27:
I don't use MoveitCommander in my code.How can I avoid segmentation fault.
This error often occurs when there are multiple orders.
This error prevented me from testing the code.

```
Segmentation fault
================================================================================REQUIRED process [gazebo-2] has died!
process has died [pid 28936, exit code 139, cmd /home/liuhaoran/ariac_ws/src/ariac-gazebo_ros_pkgs/gazebo_ros/scripts/gzserver -u -e ode -r --record_period 0.01 --record_path=/home/liuhaoran/.ariac/log/gazebo /tmp/ariac/ariac.world __name:=gazebo __log:=/home/liuhaoran/.ros/log/06df869c-dd67-11ec-8055-2cf05de1df00/gazebo-2.log].
log file: /home/liuhaoran/.ros/log/06df869c-dd67-11ec-8055-2cf05de1df00/gazebo-2*.log
Initiating shutdown!
================================================================================
```

liuhaoran1, 2022-05-27:
[performance.log](https://github.com/usnistgov/ARIAC/files/8783740/performance.log)
This is my performance.log

zeidk, 2022-05-27:
Could you also send me the yaml file you were using?

liuhaoran1, 2022-05-27:
```
# ARIAC 2022
# Trial for the robot breakdown challenge.
# The kitting robot should stop working as soon as 1 part is placed on AGV1
# To disable the gantry robot, use gantry_robot instead of kitting_robot

options:
  insert_models_over_bins: true
  belt_population_cycles: 0
  insert_models_over_stations: false
  gazebo_state_logging: true
  # mandatory: gripper_tray or gripper_part
  current_gripper_type: gripper_part
time_limit: -1

# aisle_layout:
#   person_1: # located at as2
#     location: 3
#     start_time: 0.
#     move_time: 5.
#     wait_time: 15.
  # person_2: # located at as1
  #   location: -3
  #   start_time: 16.
  #   move_time: 5.
  #   wait_time: 7.

# The table_tray_infos is not mandatory
# The different options for tray_model are:
#  -movable_tray_metal_shiny
#  -movable_tray_metal_rusty
#  -movable_tray_dark_wood
#  -movable_tray_light_wood
# quantity can be: 1, 2, or 3
table_tray_infos:
  table_2: 
    tray_model: movable_tray_light_wood
    quantity: 1

# mandatory: location of agvs and products at start
agv_infos:
  agv1:
    location: ks1
  agv2:
    location: ks2
  agv3:
    location: ks3
  agv4:
    location: ks4

orders:
  order_0:
    priority: 1
    kitting_robot_health: 1
    assembly_robot_health: 1 
    disable_robot: [assembly_robot, agv1, 1] # [assembly_robot, agv1, 1]
    announcement_condition: time
    announcement_condition_value: 0.0
    kitting:
      shipment_count: 1
      trays: [movable_tray_light_wood]
      agvs: [agv1]
      destinations: [as2]
      products:
        part_0:
          type: assembly_sensor_red
          pose:
            xyz: [-0.1, -0.1, 0.0]
            rpy: [0.0, 0.0, 0.0]
  order_1:
    priority: 1 # default is 1 if omitted, we could omit this line
    kitting_robot_health: 1 
    assembly_robot_health: 1
    announcement_condition: kitting_submission
    announcement_condition_value: [agv1, as2]
    assembly:
      shipment_count: 1
      stations: [as2]
      products:
        part_0:
          type: assembly_sensor_red
          pose:
            xyz: [0.032, -0.153, 0.076]
            rpy: [0, 0, 0]

models_over_bins:
  bin2:
     models:
       assembly_sensor_red:
         xyz_start: [0.2, 0.2, 0.0]
         xyz_end: [0.4, 0.4, 0.0]
         rpy: [0, 0, 0]
         num_models_x: 2
         num_models_y: 2

# don't forget to include at least 1 empty line here
```

liuhaoran1, 2022-05-27:
I wonder if this segmentation fault will affect the final score.I don't know how to solve.

zeidk, 2022-05-27:
Thank you, I understand the frustration. The problem is that the error can either be from your side or from Gazebo side. What is your team name? I would like to grab your code from the files you submitted on Google Drive and try it locally on my machine.

liuhaoran1, 2022-05-27:
Team Reaper

zeidk, 2022-05-27:
Is your code up to date on github? Can I clone it on my machine using build_team_system.bash?

db7296, 2022-05-27:
Most of the seg faults that I get are associated with moving the agvs.

liuhaoran1, 2022-05-28:
> 你的代码在 github 上是最新的吗？我可以使用 build_team_system.bash 在我的机器上克隆它吗？

Sorry, I will inform you about Monday when I upload the latest code.

AravindaDP, 2022-05-29:
For us also same issue occurs when picking items for an assembly shipment followed by kitting shipment (As described in point 2 under https://github.com/usnistgov/ARIAC/issues/166)

liuhaoran1, 2022-05-30:
> Is your code up to date on github? Can I clone it on my machine using build_team_system.bash?

I have uploaded the latest code on github. You can clone it on your machine using build_team_system.bash.

AravindaDP, 2022-05-31:
segmentation fault on our case is resolved with 5/30 release.

Runaround Robotics

zeidk, 2022-05-31:
Thank you for the update.

zeidk, 2022-05-31:
Closing for now.

closed 2022-05-31 by zeidk