usnistgov/ARIAC #176: submit fails on kitting agv2

db7296, 2022-05-27:
Trying to submit in non-development mode.  Works OK on briefcases and kitting agv1.  However, on kitting agv2, the server shows this error:

[ERROR] [1653626695.221936363, 87.212000000]: [ARIAC TaskManager] Service to get the content of a movable tray does not exist

zeidk, 2022-05-27:
I wonder if this is related to the issue you reported yesterday on the seg fault.

zeidk, 2022-05-27:
I found the origin of the issue you are having with agv3. I've been running your code locally trying to debug the original issue posted here. The first approach is to just spawn the parts and call the services using GearSpawner.sh but the segmentation fault never happens when shipments are submitted this way. I'm looking into the plugins related to the arms, such as the vacuumgripper plugin to see if the issue comes from there but so far nothing. This may be related to your own code but that would be tricky to debug.

db7296, 2022-05-27:
Sounds difficult.  However, nothing in the client code should be able to induce a seg fault in the server code, right?

db7296, 2022-05-27:
I'm looking into what issues in my code could be causing the submission failure.  With just a kitting order and no breakdowns, it submits ok.  If I find something, I'll let you know ASAP.

db7296, 2022-05-28:
I'm stumped.  The submission fails even on AGV1 when the assembly bot is involved but succeeds when just a kitting order is sent.  But the submissions look identical to me:

[INFO] [1653689965.020758, 129.902000]: waiting for kitting submit service /ariac/agv1/submit_kitting_shipment
[INFO] [1653689965.025717, 129.903000]: calling kitting submit service /ariac/agv1/submit_kitting_shipment, with station name = as1 and shipment name order_0_kitting_shipment_0
error when calling SubmitKittingShipment service:  service [/ariac/agv1/submit_kitting_shipment] responded with an error: 

[INFO] [1653696521.588952, 1001.265000]: waiting for kitting submit service /ariac/agv1/submit_kitting_shipment
[INFO] [1653696521.592536, 1001.266000]: calling kitting submit service /ariac/agv1/submit_kitting_shipment, with station name = as1 and shipment name order_0_kitting_shipment_0
[INFO] [1653696521.648807, 1001.268000]: submit agv kitting service result = success: True
message: ''

db7296, 2022-05-28:
OK, something really strange is going on.  Other trial configs are submitting correctly.  Perhaps there is something wrong with this trial config?
```
# ARIAC 2022
# Trial for the robot breakdown challenge.
# The kitting robot should stop working as soon as 1 part is placed on AGV1
# To disable the gantry robot, use gantry_robot instead of kitting_robot

options:
  insert_models_over_bins: true
  belt_population_cycles: 0
  insert_models_over_stations: false
  gazebo_state_logging: true
  # mandatory: gripper_tray or gripper_part
  current_gripper_type: gripper_part
time_limit: -1

# aisle_layout:
#   person_1: # located at as2
#     location: 3
#     start_time: 0.
#     move_time: 5.
#     wait_time: 15.
  # person_2: # located at as1
  #   location: -3
  #   start_time: 16.
  #   move_time: 5.
  #   wait_time: 7.

# The table_tray_infos is not mandatory
# The different options for tray_model are:
#  -movable_tray_metal_shiny
#  -movable_tray_metal_rusty
#  -movable_tray_dark_wood
#  -movable_tray_light_wood
# quantity can be: 1, 2, or 3
table_tray_infos:
  table_2: 
    tray_model: movable_tray_light_wood
    quantity: 3

# mandatory: location of agvs and products at start
agv_infos:
  agv1:
    location: ks1
  agv2:
    location: ks2
  agv3:
    location: ks3
  agv4:
    location: ks4

orders:
  order_0:
    priority: 1
    kitting_robot_health: 1
    assembly_robot_health: 1 
    disable_robot: [kitting_robot, agv1, 1] # [assembly_robot, agv1, 1]
    announcement_condition: time
    announcement_condition_value: 0.0
    kitting:
      shipment_count: 1
      trays: [movable_tray_light_wood]
      agvs: [agv1]
      destinations: [as1]
      products:
        part_0:
          type: assembly_pump_red
          pose:
            xyz: [-0.1, -0.1, 0.0]
            rpy: [0.0, 0.0, 0.0]
        part_1:
          type: assembly_pump_blue
          pose:
            xyz: [0.1, 0.1, 0.0]
            rpy: [0.0, 0.0, 0.0]
models_over_bins:
  bin1:
     models:
       assembly_pump_blue:
         xyz_start: [0.2, 0.2, 0.0]
         xyz_end: [0.4, 0.4, 0.0]
         rpy: [0, 0, 0]
         num_models_x: 2
         num_models_y: 2
  bin2:
     models:
       assembly_pump_red:
         xyz_start: [0.2, 0.2, 0.0]
         xyz_end: [0.4, 0.4, 0.0]
         rpy: [0, 0, 0]
         num_models_x: 2
         num_models_y: 2
  bin3:
     models:
       assembly_pump_red:
         xyz_start: [0.2, 0.2, 0.0]
         xyz_end: [0.4, 0.4, 0.0]
         rpy: [0, 0, 0]
         num_models_x: 2
         num_models_y: 2

# don't forget to include at least 1 empty line here
```

db7296, 2022-05-28:
I found a bug in my code, calling the wrong station name.  Sorry for the distraction.

closed 2022-05-28 by db7296

db7296, 2022-05-28:
Standby......... checking if the bug was actually causing the problem or not......