usnistgov/ARIAC #178: start point deviates from current robot state more than 0.01

AravindaDP, 2022-05-29:
Sometimes when simulation is started, kitting arm falls on conveyor belt and this causes move_it commander to fail with 'start point deviates from current robot state more than 0.01' error repeatedly. (Perhaps because conveyor belt is constantly trying to drag along the end effector of arm in this situation?)

Is there an alternate approach to get this situation resolved?

Or is it a possibility to modify competition environment so that perhaps kitting arm starts facing opposite side? (Assuming this will resolve the issue)

Similar issue is discussed in #87 but I could not come up with a suitable solution on competitor side.

zeidk, 2022-05-30:
This happens with very slow system. There is a good chance the script [gazebo_unpauser.py](https://github.com/usnistgov/ARIAC/blob/ariac2022/nist_gear/script/gazebo_unpauser.py) crashed during the start of the environment. This script will unpause the simulation after 10 s. 

If the script crashes during the start of simulation, deactivate it by commenting out this [line](https://github.com/usnistgov/ARIAC/blob/12f48c48c2d36935cf54cdafa9376a5b75e147f4/nist_gear/launch/gear.launch.template#L105). You will need to unpause the simulation yourself by clicking the play button at the bottom of the gazebo window.

If the script does not crash but you still have the same behaviors, you can increase the wait time by changing the number (from 10 to 20 for instance). 

Note that this issue you are having with the arm does not happen in a docker environment during evaluation.

zeidk, 2022-06-02:
What happens with the gantry arms? Are they hanging on the side of the torso or are they in a specific configuration? 

What you can try is to modify kitting.launch with a different configuration for the arm so it faces the other side, modify the value for  shoulder_pan_joint.

AravindaDP, 2022-06-02:
Yes gantry arms starts in a hanging configuration but that's fine since it comes to rest and can be commanded to a desired state by moveit commander interface.

Did you mean `default_initial_joint_states` for kittting arm in gear.py? What about during the competition? (I thought competitors can't change default positions themselves unless you are thinking to change it based on our feedback)

zeidk, 2022-06-02:
I can make the change in kitting.launch but I'm afraid this will affect the competitors' code. I can probably send email asking if that would cause any problem. If some competitors do not agree with the change I will run your package a few more times. Could you modify your code to print something on the screen to let me know the robot is stuck (since I have no visual during the evaluation)?

AravindaDP, 2022-06-03:
@zeidk We also agree that it's not a good idea to change the competition environment this late. 
We are good if we are granted a rerun if this happens. We'll print following using `rospy.logerr`

`Unable to move kitting arm at startup after MAX_ATTEMPTS.`

zeidk, 2022-06-03:
Thanks, that should be enough for me to understand what is happening.

AravindaDP, 2022-06-03:
I'm rather leaving this open if any other team wants to chime in whether they also come across same situation or were using different control method that is able to overcome this situation.

zeidk, 2022-06-03:
Does this issue happen at start of the simulation or when you are trying to send a command? From my understanding it is at start, is this correct?

AravindaDP, 2022-06-04:
Yes at start.

We adapted code in `kitting_commander_node.py` from test_competitor. When simulation starts, kitting arm sometimes ends up in above position as in video before competitor software has a chance to issue any commands. In this case go_home (i..e. go_to_preset_location function with kitting_robot) fails with above error repeatedly.

The same error sometimes occur during the middle of the competition as well. But it eventually succeed on 1 or 2 attempts as intended. (In this case we'll not print the error message mentioned above)

closed 2022-08-01 by zeidk