usnistgov/ARIAC #181: The Faulty Gripper challenge

liuhaoran1, 2022-05-31:
The Faulty Gripper challenge: When placing a product on an AGV, The product will fall out of The AGV.

```
drops:
  drop_regions:
    shipping_box_0_impeding:
      frame: agv1::kit_tray_1
      min:
        xyz: [-0.3, -0.3, 0.0]
      max:
        xyz: [0.3, 0.3, 0.5]
      destination:
        xyz: [0.2, 0.2, 0.05]
        rpy: [0, 0, 0.2]
      product_type_to_drop: assembly_pump_red
      robot_type: kitting # or gantry
```

zeidk, 2022-05-31:
You need to adjust the field `xyz` of the  `destination` matrix. Try `xyz: [0.15, 0.15, 0.05]` for instance or `xyz: [0.18, 0.18, 0.05]`. Decrease the `x` and `y` so that the part teleports on the AGV.

liuhaoran1, 2022-06-01:
think you

closed 2022-06-01 by liuhaoran1