usnistgov/ARIAC #185: Problem with moveit_servo dependency

JADC362, 2023-01-25:
# Problem
When installing the dependency packages of ARIAC repo, I couldn't successfully run `rosdep install` command, since `moveit_servo` is not found, even after installing all the binaries of `moveit` (`sudo apt install ros-galactic-moveit`).

JADC362, 2023-01-25:
How do you install this package? Can you do it by binary installation? Or you clone the entire repo and build it from source?

Pavel-P, 2023-01-25:
All dependencies should properly install through rosdep. Make sure you're running `sudo rosdep init` and `rosdep update --include-eol-distros` before the install step.

maxymczech, 2023-01-25:
@Pavel-P Shouldn't this information be included in the installation guide?

Pavel-P, 2023-01-25:
Yep, we'll add it next time we push an update to the documentation

PR #187 (2023-01-25): Updated installation instructions
added instructions for initializing rosdep to installation instructions

fixes #185
closed 2023-01-25 by jaybrecht