usnistgov/ARIAC #194: [Feature Req] Python test_competitor

AravindaDP, 2023-01-27:
Will it be possible to provide a python version of test_competitor for the benefit of teams using python as primarily?

It necessary not need to come up from the organizers, may be PR from a competitor already familiar with ROS2 python would also do. (Strictly restricted to the capability on par or below the level of capability in CPP version of test_competitor. After all no need to spoon feed everything)

jaybrecht, 2023-01-27:
We would like to include a python version of the test competitor but unfortunately the python bindings for MoveIt2 are not yet available. There is an [open pull request](https://github.com/ros-planning/moveit2/pull/1546) on the MoveIt2 repo that seems close to being merged. Once that is merged (and hopefully back-ported to galactic) we can explore a python version of the test competitor. 

In the meantime we welcome any pull requests from the community for a python based test competitor.

AravindaDP, 2023-01-28:
Thanks for the clarification on the blocker with moveit_py. And also, that brings up another concern (Apology on bringing it here let me know if it warrants a separate discussion) Any specific reason to stick to galactic especially now that it's at EOL? Let's hope any future enhancements to moveit_py would still be backported anyway.

jaybrecht, 2023-01-28:
Yeah the original plan with the port was to use humble but there is a bug with gazebo/ros2_control with the UR robots that makes them unusable in simulation. The issue is being tracked here: https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/issues/19

closed 2023-03-03 by jaybrecht