usnistgov/ARIAC #195: APIs

LKmubihei, 2023-02-22:
Will there be an interface to control the robot, similar to last year's "/ariac/gantry/gantry_arm_controller/command"? Even if we don't use Moveit2, we can control the robot motion through kinematics solution

jaybrecht, 2023-02-27:
There are four ROS controllers for the ceiling and floor robots that are active after you start the competition. Each of these controllers are joint trajectory controllers. If you want to control the robots without using MoveIt you can create create an action client for each of the four controllers and send [JointTrajectory goals](https://github.com/ros-controls/control_msgs/blob/master/control_msgs/action/FollowJointTrajectory.action) directly. The [controllers are detailed here](https://github.com/usnistgov/ARIAC/blob/ariac2023/ariac_description/config/robot_controllers.yaml). 

The ROS actions are named: 
```
/ceiling_robot_controller/follow_joint_trajectory
/floor_robot_controller/follow_joint_trajectory
/gantry_controller/follow_joint_trajectory
/linear_rail_controller/follow_joint_trajectory
```

LKmubihei, 2023-02-28:
Oh, thanks for the reminder, I've already tried controlling the floor robot

jaybrecht, 2023-02-28:
Also I am not sure if it says this in the documentation. I will add it if not but the controllers are inactive when the simulation starts. The competition needs to be started before the robot controllers are started and you can move the robots.

closed 2023-03-03 by jaybrecht