usnistgov/ARIAC #206: topic subscription latency

LKmubihei, 2023-03-05:
Does the Python version of the program have topic subscription latency issues? I called the self.create_client(VacuumGripperControl, "/ariac/floor_robot_enable_gripper") service, and the first time I modified the gripper state, it was successful. When I tried to change the gripper state for the second time, there was no response. So I tried to call the service in the command line, and it succeeded. Finally, I found out that the topic self.create_subscription(VacuumGripperState, '/ariac/floor_robot_gripper_state', self.state_callback, qos_profile_sensor_data) was not subscribed to the gripper state correctly.

Here is my test program：
“    interface.start_competition()
    time.sleep(1)

    print("True1,Current Gripper State",interface.gripper_enabled)
    interface.floor_robot_set_gripper_state(True)
   
    time.sleep(3)
    # rclpy.spin_once(interface)
    print("False1,Current Gripper State",interface.gripper_enabled)
    interface.floor_robot_set_gripper_state(False)

    time.sleep(5)
    interface.floor_robot_set_gripper_state(False)”

“lk@lk:~/ariac_ws$ ros2 run competition_tutorials start_competition.py 
[INFO] [1677982225.582729231] [start_competition_node]: Waiting for competition to be ready
[INFO] [1677982225.583329152] [start_competition_node]: Competition state is: ready
[INFO] [1677982225.583539762] [start_competition_node]: Competition is ready. Starting...
[INFO] [1677982225.585350468] [start_competition_node]: Started competition.
True1,Current Gripper State False
[INFO] [1677982226.587067714] [start_competition_node]: Competition state is: order_announcements_done
 msg.attached ariac_msgs.msg.VacuumGripperState(enabled=False, attached=False, type='part_gripper')
 msg.attached ariac_msgs.msg.VacuumGripperState(enabled=False, attached=False, type='part_gripper')
 msg.attached ariac_msgs.msg.VacuumGripperState(enabled=False, attached=False, type='part_gripper')
 msg.attached ariac_msgs.msg.VacuumGripperState(enabled=False, attached=False, type='part_gripper')
 msg.attached ariac_msgs.msg.VacuumGripperState(enabled=False, attached=False, type='part_gripper')
result.result() ariac_msgs.srv.VacuumGripperControl_Response(success=True)
False1,Current Gripper State False
[INFO] [1677982229.589229048] [start_competition_node]: Already disabled
 msg.attached ariac_msgs.msg.VacuumGripperState(enabled=True, attached=False, type='part_gripper')
 msg.attached ariac_msgs.msg.VacuumGripperState(enabled=True, attached=False, type='part_gripper')
result.result() ariac_msgs.srv.VacuumGripperControl_Response(success=True)
 msg.attached ariac_msgs.msg.VacuumGripperState(enabled=False, attached=False, type='part_gripper')
 msg.attached ariac_msgs.msg.VacuumGripperState(enabled=False, attached=False, type='part_gripper')
result.result() ariac_msgs.srv.VacuumGripperControl_Response(success=False)
Error calling gripper enable service
”

LKmubihei, 2023-03-05:
"Another question, have you tested the impact of multiple nodes or multi-threading on the competition? In the examples you provided, only one node is used. In ROS2, will using multiple nodes affect message reception and transmission, causing delays or other issues?"

jaybrecht, 2023-03-05:
I think the problem might be with the sleep() commands. As far as I am aware python sleep prevents all callbacks from executing properly. Can you send me the code directly to Justin.albrecht@nist.gov or post it on GitHub and I can take a look? The C++ code in the test_competitor never has any issues with latency but it is possible that python code might have problems under some circumstances. If you send me the code directly I can run some tests and try to find a solution.

LKmubihei, 2023-03-05:
I sent it to you by email. Yes, I also tested it in C++ and it runs normally. But in Python, even if I remove the sleep command, it still cannot solve the issue.

jaybrecht, 2023-03-06:
I think the issue is in how you are spinning the node. There also might have been issues since the competition was not using the simulation clock. I have created a modified version of the python competition_interface and a new executable to test changing the gripper state and it works without issues. Let me know if this code works on your end.

[competition_interface.py](https://gist.github.com/jaybrecht/4f3946754992b787664b4b2b1aea438c)
[change_gripper_state.py](https://gist.github.com/jaybrecht/31d7d517963205817aca3a6522c73327)

LKmubihei, 2023-03-06:
"Thank you, I have solved the issue with the claw. Regarding the spin function, I did have a lot of confusion about it. In the first version, I did not use the spin function and was unable to retrieve the subscribed messages from the message queue. Therefore, all the programs I wrote previously used multiple threads, where another thread was created to execute the spin function. However, after referring to your program, I did not use the spin function and was still able to receive the subscribed messages. If there is an opportunity, I hope to receive your explanation."

jaybrecht, 2023-03-06:
I would refer to the ROS documentation about how execution management works in ROS2. The examples I provided should give a starting place for competitors to build on. Closing this issue since the problem is not related to the competition.

closed 2023-03-06 by jaybrecht

zeidk, 2023-03-07:
Just one quick note about your issue. This page provides a good explanation on callback groups for rclpy https://discourse.ros.org/t/how-to-use-callback-groups-in-ros2/25255

Maybe this could be useful.

LKmubihei, 2023-03-07:
Thank you for your help, I understand a bit better now