usnistgov/ARIAC #210: Simultaneously controlling two robots

LKmubihei, 2023-03-14:
I encountered a problem while using Action communication to control the robots, which prevented me from controlling the motion of two robots simultaneously.
Here is my main code. 
`        traj = traj_generate(self.linear_joint_names,q_begin,q_end,move_time)
        self.move(self.linear_action_client,traj)
        self.wait(move_time)`
First, I calculate the trajectory information and then use Action communication in the move() function to control the robot.
Here is the implementation of the move() function. 
`    def move(self,action_client,goal_trajectory): 
        # Wait for the action server to be available
        # if not action_client.wait_for_server(timeout_sec=5.0):
        #     self.get_logger().error('Action server not available after waiting for 5 seconds')
        #     return False
        # Create a FollowJointTrajectory goal message
        goal_msg = FollowJointTrajectory.Goal()

        # Fill in the trajectory message
        goal_msg.trajectory = JointTrajectory()
        goal_msg.trajectory.joint_names = goal_trajectory.joint_names
        goal_msg.trajectory.points = goal_trajectory.points

        # Set the goal tolerance
        goal_msg.goal_tolerance = []
        goal_msg.goal_time_tolerance = Duration(seconds=5, nanoseconds=0).to_msg()

        action_client.wait_for_server()
        # Send the goal to the action server and wait for the result
        future = action_client.send_goal_async(goal_msg)
        self.goals[future] = goal_trajectory 

        future.add_done_callback(self.goal_response_callback) 

    def goal_response_callback(self, future):
        goal_handle = future.result()
        if not goal_handle.accepted:
            self.get_logger().info(f'Goal rejected')
            return
        self.get_logger().info(f'Goal  accepted')`
I suspect there is an issue with this function. At first, I tried to send the message asynchronously by using 'future = action_client.send_goal_async(goal_msg)', but this caused the trajectory messages to be overwritten, and the robot would only execute the last trajectory received. Therefore, I added the 'self.wait()' function each time I control the robot's motion to ensure that the trajectory was executed before moving on to the next one. However, the spin_once function in the wait() function prevented me from starting a new thread to control the motion of another robot simultaneously. So, I tried to send the message synchronously by using 'future = action_client.send_goal(goal_msg)', but as many tutorials mention, synchronous sending can cause deadlock.

jaybrecht, 2023-03-14:
Which action servers are you using?

jaybrecht, 2023-03-14:
I wrote a sample program for moving both robots simulataneous using action clients directly. If you try to move the robots immediately after the competition is started the robot controllers might not be started yet. I added a short wait before moving the robots to account for this issue. 

Here are two gists for reference:
[Node](https://gist.github.com/jaybrecht/fc311ebfda03e9453e569d31a4aab1a2)
[Executable](https://gist.github.com/jaybrecht/7fc0b07666497873d9ccb80416f00393)

The code was adapted from [this tutorial](https://docs.ros.org/en/galactic/Tutorials/Intermediate/Writing-an-Action-Server-Client/Py.html#writing-an-action-client) 

Closing this issue but feel free to comment if you have any questions.

closed 2023-03-14 by jaybrecht

LKmubihei, 2023-03-15:
Thank you very much for providing the code program, but it still cannot solve my problem. Now the contradiction of the problem points to action communication. The action servers I am using are the following four: "/ceiling_robot_controller/follow_joint_trajectory， /floor_robot_controller/follow_joint_trajectory，/gantry_controller/follow_joint_trajectory，/linear_rail_controller/follow_joint_trajectory".

The problem still exists that I have to add the wait() function after controlling the robot's movement each time, otherwise the robot will only execute the last planned trajectory."

 

I will provide my simple code for your reference.Below, I will use the example of replacing the gripper on the floor robot to illustrate the question.

 

The entry point of the program is Main.py.   After opening Gazebo, you can run "ros2 run action_test test" to test the code.

jaybrecht, 2023-03-15:
I don't see any code to run. But if you are saying that you need to wait for a trajectory to finish before sending a new trajectory that is correct. The FollowJointTrajectory action allows preempts the current goal when a new goal is sent. You will have to use the done_callback to check when the trajectory is finished executing before you can send the next goal. Additionally If you want to send multiple goal points as a single trajectory that can be done by adding more JointTrajectoryPoints to the FollowJointTrajectory goal before sending.

LKmubihei, 2023-03-15:
Are you saying that you can't receive my code? I've uploaded it to GitHub at https://github.com/LKmubihei/action_test/tree/main/action_test. I tried using a flag to determine whether a trajectory has been executed, but as you said, the FollowJointTrajectory action allows preemption, so only the last done_callback function will be executed. I still can't ensure that every trajectory is completed. If I wait for the execution to complete, it will cause blocking.

LKmubihei, 2023-03-17:
Excuse me, can this action communication problem be resolved? If not, I will have to control only one robot at a time to execute commands.

zeidk, 2023-03-17:
Hello, this seems to be a research question rather than a question related to the ARIAC software or API. We do not provide solutions to competitors' package as this is something competitors have to do themselves. Please let us know if you have questions specific to ARIAC. This includes bugs in the provided software or clarification on ARIAC functionalities that are not clear in the documentation. 

Also, it's not a good idea to publicly share your github repository. This is a competition and you may want to hide your code for your personal use.

Thank you.

LKmubihei, 2023-03-18:
Okay, thank you for your reply. I will take note of it.