usnistgov/ARIAC #211: Check flipped part

LKmubihei, 2023-03-21:
Check if the object needs to be flipped; the method given in the tutorial is to use a quality check. However, the quality check can only be used on the AGV, which means I need to place the object on the AGV before I can detect it. This makes the competition more complicated. Will it, like last year, indicate whether the object needs to be flipped in the order?

jaybrecht, 2023-03-22:
For kitting tasks, unless you are able to determine if a part is flipped through using a sensor the only way to determine if it is flipped is through the quality check service. For assembly/combined tasks the pre assembly poses service would allow you to determine if a part is flipped by evaluating the pose. There is no plan to tell competitors if an order includes a flipped part.

LKmubihei, 2023-03-22:
Thank you, I see it

| |
***@***.***
|
|
***@***.***
|

---- Replied Message ----
| From | Justin ***@***.***> |
| Date | 03/22/2023 21:03 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [usnistgov/ARIAC] Check flipped part (Issue #211) |

For kitting tasks, unless you are able to determine if a part is flipped through using a sensor the only way to determine if it is flipped is through the quality check service. For assembly/combined tasks the pre assembly poses service would allow you to determine if a part is flipped by evaluating the pose. There is no plan to tell competitors if an order includes a flipped part.