usnistgov/ARIAC #218: Gazebo crashes for parts on floor

jaybrecht, 2023-03-27:
Appears to be a problem with the object disposal plugin with the floor model.

jaybrecht, 2023-04-04:
@LKmubihei I just tested dropping each of the parts using the current release and I did not see any crashes. Please test on your end and let me know if the current release fixes this issue for you.

closed 2023-04-04 by jaybrecht

LKmubihei, 2023-04-06:
[image]
Dropping parts still causes Gazebo to crash, as shown in the terminal. Does your configuration require loading some online models? Some URLs are inaccessible to me.

jaybrecht, 2023-04-06:
Yes this issue seems to be related to the agent which is for the human. When you get the human working see if this bug still happens.

LKmubihei, 2023-04-17:
[image]
This issue still occurs where the dropping of parts causes Gazebo to crash.

LKmubihei, 2023-04-17:
[image]
Sometimes this also happens when a part is dropped from an AGV due to instability