usnistgov/ARIAC #219: Thread question

LKmubihei, 2023-04-04:
Is there a solution to the problem I raised at the last meeting about two robots moving simultaneously?

jaybrecht, 2023-04-04:
Please take a look at the [branch tutorial_8](https://github.com/jaybrecht/ariac_tutorials/tree/tutorial_8) on the ariac_tutorials repo. Let me know if this solves the issue.

jaybrecht, 2023-04-05:
My suggestion in this case would to have separate nodes for each robot and a single supervisor node all started separately through a launch file. The robot commander nodes would each run exclusively through callbacks and the supervisor node can send requests to the commander nodes through ROS topics, services, or actions. 

For example the floor robot commander node could host a service server called PickObjectFromBin. The supervisor could send a request asynchronously and the floor robot would execute all of the necessary movements then return a response when the request is completed. At the same time the supervisor could also send a request to the ceiling robot to perform another task through a service call or action goal. You would need to define all the necessary ROS messages yourself in this case but it should be doable. Trying to do everything in a single executable in different threads may be possible but will probably be much harder to organize and much easier to create a deadlock. 

As an example you can look at [tutorial_7](https://ariac.readthedocs.io/en/latest/tutorials/tutorial_7.html). There is the interface node that sends a Trigger request to the commander node to move each of the robots to a home position. In this case there is a single robot commander node but for more flexibility each robot could have its own separate commander node.

jaybrecht, 2023-04-05:
Also once you are in the service callback inside of one of the robot commander nodes you will need to handle sending the goal to the action server to move the robot. This [link](https://answers.ros.org/question/322831/ros2-wait-for-action-using-rclpy/) shows an example for sending a ROS action inside of service callback using the Event class as the flag to wait until the action is completed. 

Note that you need to add the service server and action client to a Reentrant callback group and start the node as part of a multi-threaded executor in order for this to work properly.