usnistgov/ARIAC #222: order  valid_id

LKmubihei, 2023-04-08:
Does this mean that each order can only invoke quality check once, but when I use it in practice, I can invoke the quality check service multiple times
[image]
The valid_id is still True
[image]

LKmubihei, 2023-04-08:
Oh, Does it mean that orders that have passed quality check will be considered invalid?

zeidk, 2023-04-09:
This means that if you call this service with a wrong order ID then valid_id returns false. Let's say you have an order with the ID "MH300" and you call this service by passing the ID "MH301", then valid_id returns false since "MH301" does not exist.

LKmubihei, 2023-04-09:
I understand what you mean, but what I want to ask is whether the quality check service can only be called once.

zeidk, 2023-04-10:
This is correct, you can call this service multiple times.

LKmubihei, 2023-04-10:
ok,thanks

anthonydowns, 2023-04-28:
Closing this issue as complete

closed 2023-04-28 by anthonydowns