usnistgov/ARIAC #230: Docker instructions

AravindaDP, 2023-04-20:
Hi,

It was announced that docker instructions would be provided on 17th but I could not locate any instructions regarding where and how to obtain base docker images and build the competitor docker images on documentation site. I see instructions related to competitor yaml file but no instructions on how to build and test a docker image based on it.

zeidk, 2023-04-21:
Hello, we are making sure that the scripts you provided properly build the container and run your nodes inside the container. We are specifically making sure the cameras work properly. We will then release the instructions to try this on your side.

AravindaDP, 2023-04-21:
Well the question specifically is as follows.

On previous years, (Specifically last year) competitors were given set of scripts that enables to download the competition-server docker image (`pull_dockerhub_images.bash`) as well as script to build a competitor docker image using submission configuration file(s) (`prepare_team_system.bash` if my memory is correct). Also were given is a script (`run_trial.bash`) to launch competition server along with the competitor container to verify that submission configuration files are error free.

Then competitors would submit the submission configuration file(s) using which organizers would build and test competitor solutions.

This time I see instructions on how to write submission configuration file which outlines the team source code, launch file and any prerequisite packages needed. What is missing is scripts that are used to build a competitor docker image using this submission configuration yaml file and then subsequently test it alongside the competion-server image. (At a minimum, competitor image building script would be required to do necessary validations on submission configuration file.) Basically, information like which base image would be used by organizers to build the competitor docker image would be needed to deduce which additional deb/pip packages need to be listed on submission configuration file and so on.

AravindaDP, 2023-04-23:
Also it seems that not having direct control over how build time commands are issued could also be a problem.

1. In previous years we had included repositories from 3rd party using wstool/git submodules. This needs commands beyond simply specifying out git repo on submission configuration file.

2. Similarly sometimes it might be necessary to install additional package by means other than simple debian packages/pip packages.

AravindaDP, 2023-04-24:
As an example, we could take https://github.com/AravindaDP/ariac_sample.git

Assumptions:
1. main Ariac repo is already checked out built and sourced in docker image.
2. deb packages and pip packages are installed before cloning this repo.

In out workflow, this repo needs to be cloned into src folder like below.
cd ~/competitor_ws/src
git clone https://github.com/AravindaDP/ariac_sample.git .
Note: using charcater "." is important at the end of above

After cloning following commands needed to execute to pull in additional git repos. (Same effect could be obtained with git submodules and recursive cloning if using wstool is not possible)
cd ~/competitor_ws
wstool update -t src

130s, 2023-04-25:
I'm adding questions regarding Docker instruction to this ticket to not scatter questions at multiple places:

3. (Related to bullet#2 in https://github.com/usnistgov/ARIAC/issues/230#issuecomment-1518933077) Is there a way to add a custom `apt` source (i.e. the one gets added under `/etc/apt/sources.list.d`)?
4. Is there a way to just submit a [Dockerfile](https://docs.docker.com/engine/reference/builder/)? This could eliminate many concerns regarding the organizer having to build a Docker image (well, if there's no way to do this, I understand as it's too late this year).

AravindaDP, 2023-04-25:
Also appreciate if build scripts (Even though they are not finalized at this point) be shared so we could have a better understanding and raise the questions/suggest workarounds.

130s, 2023-04-25:
5. Which shell file to source? ROS2 relies on "source"ing `bash` file (e.g. `/opt/ros/galactic/setup.bash`). Is that what the competition sources?
6. Related to 5, where are competitors packages supposed to be installed? Under `/opt/ros/galactic` (so that sourcing `setup.bash` from there enables our packages without custom modification)?

Sorry for asking many questions in the last minute, but for a newbie competitor there are too many lines to read in between.

LKmubihei, 2023-04-26:
Is the release of Docker delayed, I haven't seen any updates on this in the past few days

AravindaDP, 2023-04-26:
Unfortunately, no additional information is released than partial information available at https://ariac.readthedocs.io/en/latest/automated_evaluation/docker_setup.html released yet.

I suggest that Qualification deadline be pushed back atleast till EoD 5/1 due to following reason. (And also rectified the same at the earliest) I think all competitors would agree since these are blockers faced by all teams.

Docker build script that takes submission configuration file described in above link and builds a docker image based on it is not released yet and this is required for competitors to validate the submission configuration file. This is a critical blocker still not resolved.

anthonydowns, 2023-04-26:
Hello all, I will send out an email on this when I get a chance to confirm with my team, but we see the concerns and will most likely push back the deadline for the Qualifier Submissions to be due.

anthonydowns, 2023-04-26:
Hello all, again, I will send an announcement about this, but we will be pushing the deadline back to at least COB Monday. Further information will be coming out in the next day or so.

anthonydowns, 2023-04-28:
The deadline is now next Friday. And I believe there should be a merge happening later today that will give the answers to at least some of these issues. Thanks for your patience.

AravindaDP, 2023-05-02:
Scripts related to Automated Evaluation procedure is available. However, documentation related to the same is not updated yet as indicated in https://ariac.readthedocs.io/en/latest/automated_evaluation/docker_setup.html#automated-evaluation-procedure

From the latest update I see following deficiencies in the approach.

1. deb install list provided in submission configuration file does not seem to be used in https://github.com/usnistgov/ARIAC/blob/ariac2023/automated_evaluation/autoEval/read_competitor_config.py
2. Additionally competitors might need ability to run additional prebuilt script (To pull in additional 3rd party packages distributed in source form etc.) This could be achieved by including additional field under competition called `prebuild_script_file` (similar to `launch_file`) and if it is not empty, launch it in a subprocess from `~/ariac_ws/src/{team_name}/{prebuild_script_file}` location. (With working directory ideally set to `~/ariac_ws/src/{team_name}` folder. But any other predefined location also would work as long as it is clearly defined in documentation) This command should be ideally placed just before https://github.com/usnistgov/ARIAC/blob/ariac2023/automated_evaluation/autoEval/read_competitor_config.py#L30 (rosdep install step. But it could be anywhere after clone step)

zeidk, 2023-05-02:
Hello, we are working on the automated documentation for competitors describing how to run the script. Can you make any changes that will work for you and send it back to me? It seems that your case is the most complex one and that will be a good way for us to include complex installation instructions.

AravindaDP, 2023-05-03:
Is documentation on automated evaluation is completed?

Was trying to run automated evaluation based on https://github.com/usnistgov/ARIAC/blob/documentation_staging/docs/automated_evaluation/docker_setup.md But had following issues.

1. `docker image pull jfernandez37/ariac` failed since there is no tag named `latest` But could get the image using `docker image pull jfernandez37/ariac:ariac_latest`
2. https://github.com/usnistgov/ARIAC/blob/documentation_staging/automated_evaluation/runDockerTest.sh#L2 fails since when we use it seems to use full name. So image name needed to be changed to `jfernandez37/ariac:ariac_latest`
3. Copy of the ariac_ws in docker image seems to be outdated. So it does not have automated_evaluation folder in it. So command at https://github.com/usnistgov/ARIAC/blob/documentation_staging/automated_evaluation/runDockerTest.sh#L5 fails

AravindaDP, 2023-05-04:
Based on this I created https://github.com/usnistgov/ARIAC/pull/235 to install debian_packages and also ability to run bash script from competitors repository before running colcon build

LKmubihei, 2023-05-05:
Will the detailed tutorial for Docker be released in the next two days? I tried running it based on [https://github.com/usnistgov/ARIAC/blob/documentation_staging/docs/automated_evaluation/docker_setup.md ](https://github.com/usnistgov/ARIAC/blob/documentation_staging/docs/automated_evaluation/docker_setup.md ) but encountered some key-value mismatch issues and couldn't run bash script.

130s, 2023-05-06:
Submission process is NOT officially uploaded on the official page. Some folks including me on this thread dug and made their way. I guess technically eval process can be fixed on the evaluators' side even after submission due, but it's very inconvenient to NOT be able to verify the process before submission. So in the worst case if the submitted program failed during the formal evaluation, it'd be nice to see some grace (feedback and re-submission etc.) Just my thoughts.

Unless you have offline channel with the officials, the following may be the thought process each contestant may take:
1. [Submission extended to 5/7](https://groups.google.com/u/1/a/list.nist.gov/g/ariac2023/c/yJP9sZf0NBo)
1. https://ariac.readthedocs.io/en/latest/automated_evaluation/docker_setup.html#automated-evaluation-procedure reads 5/1, which way passed due.
   > Documentation on the automated evaluation procedure will be provided on 05/01/2023.
1. Wait, there's an unmerged branch whose name reads [documentation_staging](https://github.com/usnistgov/ARIAC/tree/documentation_staging) thanks to https://github.com/usnistgov/ARIAC/issues/230#issuecomment-1533588323. There's this file [docker_setup.md](https://github.com/usnistgov/ARIAC/blob/documentation_staging/docs/automated_evaluation/docker_setup.md) so follow that.
1. It fails early on. I guess everyone is making their own fixes locally?
   - I  tried to contribute fixing the process, which I submitted https://github.com/usnistgov/ARIAC/pull/236.
   -  I also see another PR https://github.com/usnistgov/ARIAC/pull/235 that seems to aim fixing the eval process.

jaybrecht, 2023-05-11:
Hopefully all confusion and questions will be solved with v1.4 and the new [instructions](https://ariac.readthedocs.io/en/latest/automated_evaluation/automated_evaluation_overview.html). If not please post on this [discussion page](https://github.com/usnistgov/ARIAC/discussions/242). Closing this issue in favor of the discussion page.