usnistgov/ARIAC #232: faulty_part

LKmubihei, 2023-04-27:
Regarding faulty_part, it is set in the faulty_part challenge
But in the sample.yaml file, there are faulty parts too
[image]

 Does this mean that faulty_part could also appear on the bin?

anthonydowns, 2023-04-28:
A faulty part can be any part in the environment, but you will only be notified that it is faulty when it goes through the quality control sensor on the AGV (I believe)

zeidk, 2023-05-02:
Hello, 
The screen capture you posted seems to be from an older yaml file that we forgot to update. We are not using this way to set faulty parts in the environment.

The only way to set  faulty parts is with the ```challenges``` field. Here is an example:

```yaml
challenges:
  - faulty_part:
      order_id: 'MMB30H00'
      quadrant1: false
      quadrant2: true
      quadrant3: true
      quadrant4: true
```

LKmubihei, 2023-05-02:
I see it, Thank you

closed 2023-05-11 by jaybrecht