usnistgov/ARIAC #244: CHECK failed: (index) < (current_size_) error in libprotobuf causes gazebo crash when FlexBe is running.

AravindaDP, 2023-05-13:
I'm currently having gazebo crashing at random times with following error.
```
[gzserver-2] [libprotobuf FATAL /usr/include/google/protobuf/repeated_field.h:1506] CHECK failed: (index) < (current_size_):
[gzserver-2] terminate called after throwing an instance of 'google::protobuf::FatalException'
[gzserver-2]   what():  CHECK failed: (index) < (current_size_):
[gzserver-2] Aborted
[ERROR] [gzserver-2]: process has died [pid 24947, exit code 134, cmd 'gzserver /home/aravindadp/ariac_ws/install/ariac_gazebo/share/ariac_gazebo/worlds/ariac.world
                  -s libgazebo_ros_init.so   -s libgazebo_ros_factory.so   -s libgazebo_ros_force_system.so       '].
```

Further following were observed.
1. In fact simply using FlexBe do not cause the issue. But only when Subscriber was created in FlexBe behavior to read `/ariac/competition_state`
2. Tried republishing `/ariac/competition_state` into different topic using ros `topic_tools` `relay` node and then using that topic instead `/ariac/competition_state` but still problem occurs. (Note: simply relaying to a new topic using topic tools did not caused any issue.)
3. In most cases issue occurs while robots are moving. It occurs both when something is attached to gripper or even when gripper is empty)

Following repo can be used to recreate the issue.
https://github.com/AravindaDP/ariac_sample
Note: after cloning the repo following command need to be executed inside repo (where .rosinstall file exists) to checkout additional repos.
sudo apt-get install python3-wstool
wstool update -t .

after installing dependancies using rosdep and colcon build following command runs test_competitor along with competiton and flexbe
`ros2 launch arac_competitior arac_competitior.launch.py`

It's not clear if it was due to a issue in FlexBe cached subscriber implementation is causing this issue or not (Since FlexBe port of ros2 is still not fully completed and might have some unfixed issues)

While organizers may not be able to guarantee compatibility with every ros packages out there, posting here with as much as information in case someone else comes with similar issue or be able to shed more light on what could be actually causing the issue.

LKmubihei, 2023-05-15:
Sometimes I make the same error.

AravindaDP, 2023-05-16:
Potentially this issue is occured due to extreme rate of publishing of following topics
```
/ariac/competition_state
/ariac/robot_health
/ariac/sensor_health
```
There may be other topics as well that are getting published at each simulation update step (1kHz)

closed 2023-05-22 by jaybrecht

LKmubihei, 2023-05-26:
[image]
Crashes still occur when I test assembly.yaml, which happens occasionally