usnistgov/ARIAC #249: /ariac/perform_quality_check returns incorrect response when quadrant3 is faulty

AravindaDP, 2023-05-21:
When following challenge is used with kitting.yaml task_manager plugin detects that quadrant3 is faulty But /ariac/perform_quality_check service returns that all are good.

```
challenges:
  - faulty_part:
      order_id: 'MMB30H56'
      quadrant3: true
```

```
[gzserver-2] [INFO] [1684665740.157941197] [task_manager]: Part in quadrant 3 is faulty
[gzserver-2] [INFO] [1684665740.161510020] [agv4_plugin]: Moving agv4 warehouse
```
...
```
[gzserver-2] Quadrant 1 score: 3
[gzserver-2] Quadrant 2 score: 0
[gzserver-2] Quadrant 3 score: 0
[gzserver-2] Quadrant 4 score: 0
```
[faulty_part.yaml.txt (renamed, remove .txt extension)](https://github.com/usnistgov/ARIAC/files/11524310/faulty_part.yaml.txt)

AravindaDP, 2023-05-22:
Can confirm that issue249 branch fixes issue in this case.

On a side note, inspecting issue249.yaml (And inability of test_competitor to handle it) raise the question whether there could be situations of partially completed orders at the time of announcement? If my memory was correct, in fact previous years, at least the tray could have been already populated. (Parts may be but I thought that option wasn't exercised)

If we should check for the partially completed state, should it also include having to remove any unwanted parts and or rearranging parts in case in wrong positions?

jaybrecht, 2023-05-22:
The issue249.yaml file was just for testing without having to wait for the test competitor to fill out the tray. No trays will be partially filled for non-assembly order for the competition.

closed 2023-05-22 by jaybrecht