usnistgov/ARIAC #252: For flipped part (pump) vacuum_gripper_plugin detects attachment but gazebo do not pick up object.

AravindaDP, 2023-05-21:
Following is logged in output but pump is not picked up in simulation as usual.
```
[gzserver-2] [INFO] [1684679930.392990118] [floor_vacuum_gripper_plugin]: Picking object
```

jaybrecht, 2023-05-22:
Can you recreate this and record your screen for reference? If the simulation posts this message than the joint between the robot and the part has already been established so it should be attached at this point.

jaybrecht, 2023-05-22:
Closing in favor of #251

closed 2023-05-22 by jaybrecht

AravindaDP, 2023-05-22:
Actually it seems the issue is somewhere else. It seem to be incorrect planning scene handling in competitor code for the case without considering that part links are on bottom of the part instead in the middle.

jaybrecht, 2023-05-22:
The part links are on the bottom of the part though.