usnistgov/ARIAC #253: Result feedback

LKmubihei, 2023-05-24:
The announcement says that the results of the qualifiers will be fed back to Google Drive on May 23, but I haven't received any feedback yet

anthonydowns, 2023-05-25:
Team Reaper, please check to make sure that you have your files in your folder. It looks empty on our end. Nothing for us to run.

jaybrecht, 2023-05-25:
I see the submission now, I am not sure why we couldn't see it before. I am running the qualifier trials for your team now. It does seem like you are not ending the competition after all orders are submitted. This is not necessarily a problem the program will run until the trial time limit but it would be better for you to end the competition through the ROS service when your program finishes all orders.

AravindaDP, 2023-05-25:
Team Runaround Robotics also have files in drive from 23rd (Note that we also have pre-build script in competitor_build_scripts folder)

LKmubihei, 2023-05-25:
[image]
Does Ti mean from the beginning of the competition to the end of the competition? That said, I need to end the competition myself to reduce the time consuming. 
The reason I didn't end the competition before was that I didn't know when the order would come, if I end the competition after submitting the first order, but at this time the second order had not been announced. For example, this problem occurs in the human_kitting task.the second order was announced at 60s, but my first order was submitted before 60s

AravindaDP, 2023-05-25:
I see some files with high level results. Will it be the same level of information we'll be receiving in the finals as well?
1. Can we have full roslog/console output?
2. Will it be possible to have gazebo state log and script that can be used to replay similar to last year?
It's OK if you don't have facility to provide these at the moment and looking forward to implement it for the final scoring.

LKmubihei, 2023-08-10:
Hi, the results of this year's competition seem to be a bit slow to be released, I was wondering when the results will be announced?

zeidk, 2023-08-12:
Hello,

We will announce the results as soon as we have the results from the human judges. Hopefully next week. We apologize for the delay.

closed 2023-11-27 by jaybrecht