usnistgov/ARIAC #255: Docker image outdated?

AravindaDP, 2023-06-05:
Started getting bunch of errors on docker image build. Could it be outdated and no longer able to install packages from apt due to this?

jaybrecht, 2023-06-05:
Could you try updating the token with the latest from step 9 of Testing the Automated Evaluation on this page: https://ariac.readthedocs.io/en/latest/automated_evaluation/automated_evaluation_overview.html

AravindaDP, 2023-06-05:
It is already in that latest token. Besides same error for our own repo. And also apt-get update and apt-get installs inside docker is also causing (different to above though) errors. Not sure if it's a problem of my local docker setup but it was working without any issues earlier. May be some certificate/key file expired in underline docker image?

AravindaDP, 2023-06-05:
OK. Then may be it's a issue in my local setup. If our image is properly building (runaround_robotics) it should be good to go.

closed 2023-06-05 by AravindaDP