usnistgov/ARIAC #289: seg fault from plan_and_execute method

dan9thsense, 2024-01-18:
Calling the tutorial 10 competition interface method _plan_and_execute (generally without plan_parameters) multiple times eventually (after about 6 calls) generates a segmentation fault.  There is no additional information in the logs about what generates it.

process has died [pid 41695, exit code -11,

jfernandez37, 2024-01-22:
Here is the link to the new tutorials and the documentation for it. The tutorial you are running is out of date. Hopefully, these work better.
Tutorials: [https://github.com/usnistgov/ARIAC_tutorials.git](https://github.com/usnistgov/ARIAC_tutorials.git)
Documentation: [https://pages.nist.gov/ARIAC_docs/en/latest/tutorials/tutorials.html](https://pages.nist.gov/ARIAC_docs/en/latest/tutorials/tutorials.html)

dan9thsense, 2024-01-23:
OK, thanks for the update.

closed 2024-01-23 by dan9thsense