usnistgov/ARIAC #293: pump goes crazy when trying to attach

dan9thsense, 2024-01-25:
When slowly lowering down to a battery, things work fine, but slowly lowering onto a pump makes it go flying all about as soon as contact is made.  Perhaps there is a collision bug?

jaybrecht, 2024-01-25:
PR #288 added some changes that should make assembly and part picking more reliable/stable. Are you working with the most recent version? 

Also are you using the tutorials package (python) or the test competitor (c++)?

dan9thsense, 2024-01-25:
Good call.  Pulling the new version fixed the problem.

closed 2024-01-25 by dan9thsense

jaybrecht, 2024-01-25:
Glad that it is fixed. What is currently on the ariac2024 branch (renamed from ariac2024_dev) has all of the changes we are planning for this year's competition . We will add any bug fixes necessary but there shouldn't be any more large changes or features.