usnistgov/ARIAC #297: service call /ariac/get_pre_assembly_poses ariac_msgs fails

dan9thsense, 2024-01-27:
This service call to get the pre_assembly poses becomes available but the actual call either times out or always returns valid_id=False, even with a correct order_id.

ros2 service call /ariac/get_pre_assembly_poses ariac_msgs/srv/GetPreAssemblyPoses "order_id: 'MMB30H56'"
waiting for service to become available...
requester: making request: ariac_msgs.srv.GetPreAssemblyPoses_Request(order_id='MMB30H56')

response:
ariac_msgs.srv.GetPreAssemblyPoses_Response(valid_id=False, agv_at_station=False, parts=[])

jaybrecht, 2024-01-27:
Have you moved the AGV/AGVs to the station for the order? I believe that is why it is not working.

jaybrecht, 2024-01-27:
Actually I just checked the code. It should still return valid id as true even if the AGVs are not at the station. Which trial are you testing with?

dan9thsense, 2024-01-27:
I'm using the assembly.yaml trial and was able to get the order_id to return True with the command line call. 

I have not moved the AGVs yet.  "agv_at_station" reports False correctly.  "parts" does not report any partposes.  Do the AGVs need to be in place to get partposes?

ros2 service call /ariac/get_pre_assembly_poses ariac_msgs/srv/GetPreAssemblyPoses "order_id: '2IZJP127'"
waiting for service to become available...
requester: making request: ariac_msgs.srv.GetPreAssemblyPoses_Request(order_id='2IZJP127')

requester: making request: ariac_msgs.srv.GetPreAssemblyPoses_Request(order_id='2IZJP127')

response:
ariac_msgs.srv.GetPreAssemblyPoses_Response(valid_id=True, agv_at_station=False, parts=[])

dan9thsense, 2024-01-27:
I moved the AGVs and now it all works fine.  Sorry to bother you about this one-- should have thought of that!

closed 2024-01-27 by dan9thsense