usnistgov/ARIAC #307: ariac.launch.py fails due to gazebo not being sourced

endre90, 2024-02-06:
#### OS:
Ubuntu 22.04
#### ROS2:
Humble (binaries)
#### Issue:
```
ros2 launch ariac_gazebo ariac.launch.py
...
[gzserver-1] [INFO] [1707231972.413343279] [task_manager]: Starting ARIAC 2024
[gzclient-2] gzclient: /usr/include/boost/smart_ptr/shared_ptr.hpp:728: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::rendering::Camera; typename boost::detail::sp_member_access<T>::type = gazebo::rendering::Camera*]: Assertion `px != 0' failed.
[spawner-14] [INFO] [1707231973.849087385] [agv2_controller_spawner]: Waiting for '/controller_manager' node to exist
[spawner-8] [INFO] [1707231973.850133767] [joint_state_broadcaster_spawner]: Waiting for '/controller_manager' node to exist
[spawner-16] [INFO] [1707231973.862401182] [agv4_controller_spawner]: Waiting for '/controller_manager' node to exist
[spawner-11] [INFO] [1707231973.864487873] [linear_rail_controller_spawner]: Waiting for '/controller_manager' node to exist
[spawner-15] [INFO] [1707231973.871183656] [agv3_controller_spawner]: Waiting for '/controller_manager' node to exist
[spawner-12] [INFO] [1707231973.871668267] [gantry_controller_spawner]: Waiting for '/controller_manager' node to exist
[spawner-13] [INFO] [1707231973.871852425] [agv1_controller_spawner]: Waiting for '/controller_manager' node to exist
[spawner-10] [INFO] [1707231973.878488192] [ceiling_robot_controller_spawner]: Waiting for '/controller_manager' node to exist
[spawner-17] [INFO] [1707231973.906133894] [floor_robot_static_controller_spawner]: Waiting for '/controller_manager' node to exist
[ERROR] [gzclient-2]: process has died [pid 91785, exit code -6, cmd 'gzclient'].
...
```
#### Fix that worked for me:
Additionally to sourcing the main ROS setup from `/opt/ros/` and from the `~/ariac_ws/`, source the source Gazebo's setup file:
```
. /usr/share/gazebo/setup.sh
```
P.S. Info could be added to `https://pages.nist.gov/ARIAC_docs/en/latest/getting_started/installation.html`

jaybrecht, 2024-02-06:
It should not be necessary to source gazebo's setup file in order to run the competition. I believe this error may also be due to running on humble but I am not positive. Please let me know if you continue to run into this error when running on iron.

endre90, 2024-02-07:
Thanks, I will. I'll switch to Iron now and try it out.

prasun2712, 2024-02-07:
> It should not be necessary to source gazebo's setup file in order to run the competition. I believe this error may also be due to running on humble but I am not positive. Please let me know if you continue to run into this error when running on iron.

I am facing the same issue in Iron. 
[image]

endre90, 2024-02-07:
I ran now with Iron and I have no issues.

closed 2024-02-07 by jaybrecht