usnistgov/ARIAC #309: conveyor state clarification

dan9thsense, 2024-02-15:
If the conveyor initial state has enabled=False, does that mean the conveyor will not be used for that trial?  If it is enabled, does the power turn on and off and on again during the trial?  Does the conveyor parts message remain constant throughout a trial?

jaybrecht, 2024-02-19:
- The `enabled` field indicates if the conveyor is currently moving. For the current version of the competition, the conveyor will turn on when the competition is started and remain on at full power until the end of the competition. 
- Due to the conveyor not starting until the competition is started the initial state of topic may be false even when the conveyor is going to be active for that trial, depending on when you initialize the subscriber. 
- Yes, the conveyor parts message is constant throughout a given trial. If the `parts` field is empty you can assume the conveyor is not in use for that trial.

dan9thsense, 2024-02-20:
Thanks for the clarification.

closed 2024-02-20 by dan9thsense