usnistgov/ARIAC #310: Robot state publisher log large amount of "moved backward in time" warnings.

mqt0029, 2024-02-16:
On both containerized environment and direct ROS 2 Iron installation on a fresh Ubuntu 22.04 Jammy installation, when launching ARIAC with `ros2 launch ariac_gazebo ariac.launch.py`, I am seeing a constant stream of warnings:
```
[robot_state_publisher-7] [WARN] [1708121919.739176014] [robot_state_publisher]: Moved backwards in time, re-publishing joint transforms!
[robot_state_publisher-7] [WARN] [1708121919.840994544] [robot_state_publisher]: Moved backwards in time, re-publishing joint transforms!
[robot_state_publisher-7] [WARN] [1708121919.939366885] [robot_state_publisher]: Moved backwards in time, re-publishing joint transforms!
[robot_state_publisher-7] [WARN] [1708121920.039513745] [robot_state_publisher]: Moved backwards in time, re-publishing joint transforms!
```

Does ARIAC dev team have any pointer as to why this might be happening? I have also attached a full launch log for context.

> :spiral_notepad: [**launch_log.txt**](https://github.com/usnistgov/ARIAC/files/14316534/launch_log.txt)

jaybrecht, 2024-02-19:
My best guess is that there is another node on your system that is publishing TF data/joint state data that was not properly exited. Try running

`ros2 daemon stop`

then

`ros2 daemon start`

then restarting the competition.

mqt0029, 2024-02-19:
Thanks for the suggestion, but that wasn't it.

As it turns out, I copy-pasted an earlier `docker run` snippet from old code, which has the flag `--network host` on it. While sharing the docker image between the team members, we have a couple of instances of ARIAC on various machines, all has `--network host` in it. Removing the flag solved the problem as the container no longer sees other ARIAC instances topics. The simulator started right up, with no errors.

**TL;DR: Be careful with Docker's `--network host` flag!**

closed 2024-02-19 by mqt0029