usnistgov/ARIAC #312: conveyor part wobble

dan9thsense, 2024-02-27:
When attaching the gripper to parts on the conveyor belt, they go all wobbly and crazy.  This looks similar to what the pump was doing in bins earlier, before the fix.  In this case, it does not matter what part type is being attached.  Also, it does not seem to be related to dragging the part on the conveyor, as just putting the gripper in front of a part shows that the parts can slide around on the conveyor OK.  Also, it happens even when the part is very gently approached for attaching.

[ariac_conveyor_wobbly_part.webm](https://github.com/usnistgov/ARIAC/assets/1042607/61e706b4-7449-429b-b494-bf04001e6e61)

[ariac_conveyor_wobbly_part_end_view.webm](https://github.com/usnistgov/ARIAC/assets/1042607/a4066513-bbbb-4c4b-b6ab-40fe1dd72e56)

jaybrecht, 2024-02-27:
Have you tried decreasing how far down you move the gripper when picking from the conveyor? From the videos it seems like the robot is moving down quite a bit past the surface of the part. This may not be an issue when picking from the bins since there is no side force but it is more of an issue for the conveyor parts. Also I would recommend moving up immediately after the part is successfully attached to prevent any other weird physics bugs.

dan9thsense, 2024-02-28:
I tried going as slow as possible; much slower than when picking from a bin.  I think that there is a delay in updating the gripper attached state, which causes the arm to continue to move a bit past the point where it attaches.  So far, I have not succeeded in attaching to a conveyor part without it going crazy.  I'll try some more with stops at the expected part height.  However, I'm concerned about running out of conveyor length, as the very very slow movements and stops take a lot of time.

I am unable to move up immediately after attaching because the crazy wobble moves the arm enough that the controller refuses to respond, as the expected starting joint positions are too far away.  Is there a way to ignore that and get the arm to move anyway?

jaybrecht, 2024-02-28:
I don't think speed is the issue. Can you explain a bit about how you are generating the trajectory for picking the part? 

In the nist_competitor the logic consists of planning a trajectory down to the part surface with the gripper enabled, then moving back up. After the move up we query the gripper state topic to see if the pick was successful, if not we try another pick. This is not the only possible solution but it does work fairly well and does not cause any physics bugs.

From what you have said it seems like you are planning the trajectory down past the part top surface and then stopping execution after the part is attached? Is that correct?

dan9thsense, 2024-02-28:
Yes, that approach does work but it is not something we use with our robots.  It is not an "agile" solution.  That's because it involves making relatively large arm moves (~ couple cm) to arrive at a location with high accuracy (~ couple mm).  This fails in the real world due to a lot of things, in particular gear backlash and variance in the height of an individual part or location on the conveyor (conveyor belts can sag). Moving down slowly to a point slightly below where you should get attachment works much better, as the hard stop point is high enough to not damage the part in the case of a failed gripper, yet low enough to almost always get a good attachment.

ARIAC is very helpful in robot development and I try to only use techniques that I can translate to real robots.

However, all simulations have their limitations and I can accept that the sensitivity of the conveyor parts to wild wobbles is one of those here.  Just having conveyor parts at all is a very nice feature of ARIAC. I understand the difficulty in creating it and very much appreciate that it is there and am happy to work around it.  I wanted to be sure that it was not a bug like the earlier case with the pump in the bins.

dan9thsense, 2024-02-29:
It looks like the suction from attaching actually pulls the arm down a bit as well as pulling the part up.

dan9thsense, 2024-02-29:
OK, with some tweaks and twists, it is working reliably for now.  But it is an ugly method!  Nevertheless, it works for now and I'll give it more thought in the future.  Thanks for the feedback on my question.

closed 2024-02-29 by dan9thsense