usnistgov/ARIAC #315: details on orders

dan9thsense, 2024-03-18:
Is there a maximum number of orders per run?

If a high priority order is issued, does it have to be submitted as the next issue or can a regular priority order be submitted next, accepting that the resultant overall score will likely be lower due to the priority order being delayed a bit and the priority multiplier? 
If the high priority order does need to be the next order submitted, what are the penalties if it is not?  Do both the regular and priority order scores get set to zero?

zeidk, 2024-03-18:
Hi Dan, there is no maximum numbers of orders per run.
The only thing that will happen if the high-priority order is not submitted next is the multiplier being set to 1 instead of 3 (https://pages.nist.gov/ARIAC_docs/en/latest/evaluation/scoring.html#trial-score)

You will not get a score of 0 in the example you provided.

dan9thsense, 2024-03-19:
Sounds good, thanks for the clarification.

dan9thsense, 2024-03-21:
Is there any indication that there will be no more orders issued?  Without that, we cannot safely end the competition before the full allowed time has passed.  It would be nice to be able to end the competition after all the orders are filled rather than having to sit idle on the chance that another order will be announced.

jaybrecht, 2024-03-23:
Yes the competition state topic will publish `ORDER_ANNOUNCEMENTS_DONE` after all orders have been announced.

[documentation](https://pages.nist.gov/ARIAC_docs/en/latest/competition/overview.html#program-flow)

dan9thsense, 2024-03-25:
ah, missed that, thanks.

closed 2024-03-25 by dan9thsense