usnistgov/ARIAC #316: trays for combined tasks

dan9thsense, 2024-03-18:
Are trays required on the AGV for combined tasks? If so, since a tray id is not specified, how to avoid using a tray that may be needed for the next task (possibly a high-priority kitting task)? If not, will the parts still lock down?

jaybrecht, 2024-03-18:
Trays are not explicitly required but they are recommended for combined orders because they have increased friction properties which prevent the parts from sliding when the AGV is moved. I had originally intended trays with id of 0 to be used for assembly and combined orders to fix this issue but we never actually codified that in the documentation. I will discuss with the team and get back to you. If we do make this distinction there will always be at least one tray with id of 0 if there is a combined order and that id will never be used for kitting orders.

dan9thsense, 2024-03-19:
sounds good.

closed 2024-03-19 by dan9thsense