usnistgov/ARIAC #318: assembly battery z position

dan9thsense, 2024-03-21:
In placing the battery in the assembly station, the box constrains the z position (the battery sits in the tray).  However, in scoring, the z position is seen as incorrect. In this example, the order specified z: 0.043.  However, with the battery placed in its box at the assembly station, the result reports that z is 0.040. Ignore that the y position is incorrect-- that is determined by the arm placement-- the battery was sitting flat in its box.

order:
overall.py-1] Part: 🟥 Red Battery
[overall.py-1]   Position:
[overall.py-1]     x: -0.150 (m)
[overall.py-1]     y: 0.035 (m)
[overall.py-1]     z: 0.043 (m)
[overall.py-1]   Orientation:
[overall.py-1]     roll: 0°
[overall.py-1]     pitch: 0°
[overall.py-1]     yaw: 90°
[overall.py-1]   Install direction:
[overall.py-1]     x: 0.0
[overall.py-1]     y: 1.0
[overall.py-1]     z: 0.0

result:
[gzserver-1] 	Part: red battery
[gzserver-1] 		Part score: 0
[gzserver-1] 		Correct type: yes
[gzserver-1] 		Correct color: yes
[gzserver-1] 		Correct pose: no
[gzserver-1] 			Position: -0.150051, 0.009746, 0.039655
[gzserver-1] 			Orientation: -0.001855, 0.005000, 1.573265

jaybrecht, 2024-03-23:
Do you have a picture of what the battery looks like in the insert when this occurs? Is the battery marked as assembled on the topic `/ariac/assembly_insert_{n}_assembly_state`?

dan9thsense, 2024-03-24:
battery_attached: false
pump_attached: false
regulator_attached: true
sensor_attached: false

[image]

dan9thsense, 2024-03-24:
[image]

jaybrecht, 2024-03-24:
I think the battery just needs to be installed further into the slot. If the topic is reporting false for the part it has not been recognized as "assembled" yet and you will not receive points. The assembly plugin is looking for contact between the part and a contact sensor embedded into the insert. If that contact has not been made it doesn't think the part is assembled even if it is very close to the correct pose.

dan9thsense, 2024-03-25:
Yes, that's true.  When further inserted, it scores correctly.  Mostly, I was just wondering why the z position was not the same in the score report and the order request, given that the battery sits on the floor of its holder.

closed 2024-03-25 by dan9thsense