usnistgov/ARIAC #322: seg fault with gripper service

dan9thsense, 2024-04-06:
I am occasionally (about 1 time in 8) seeing seg faults occur when I call the enable_gripper services.  I'm not sure that is the cause of the seg faults, but it is the closest thing I can see right now.  Still, it may still be my code somewhere.  Just wondering is anyone else seeing this?

dan9thsense, 2024-04-09:
There are a lot of logs. Which ones do you want to see?

jaybrecht, 2024-04-09:
Ideally the output from the terminal that you launch the ariac environment and the output from the terminal that you launch your competitor node.

dan9thsense, 2024-04-11:
Those logs don't show much.  The environment does not show the seg fault, only the competitor node.  And all that shows is "exit code 11" which is a seg fault. That's unusual in Python!  I'm concerned that the MoveIt2 C++ wrappers have some sort of deep bug.

dan9thsense, 2024-04-12:
It's looking like the cause is a bug in MoveIt2. Closing for now.

closed 2024-04-12 by dan9thsense

jaybrecht, 2024-04-12:
We will run each trial multiple times so if the bug is infrequent it shouldn't affect your final score.