usnistgov/ARIAC #323: Docker is running with an error

Zhangjiyuan1, 2024-04-11:
I got an error when following the tutorial to the steps below：
    ./run_trial.sh nist_competitor kitting
The following error message is displayed：
    Package_name:  competition_tutorials
    Launch_file:  start_competiton.launch.py
    file 'start_competiton.launch.py' was not found in the share directory of package 'competition_tutorials' which is at '/workspace/install/competition_tutorials/share/competition_tutorials'
The above launch file works fine on my own computer, but I don't know why the error occurs on docker, I guess the normal file directory should be /workspace/install/competition_tutorials/share/competition_tutorials/launch instead of the one with the error above.
I came step by step according to the tutorial on the official website, how can I solve this problem?

jaybrecht, 2024-04-11:
Are you trying to use the nist_competitor package or your own package? The name of the configuration yaml file and the name of the competitor package needs to be the same in order for everything to work properly.

Zhangjiyuan1, 2024-04-12:
My problem solved, thank you very much！

dan9thsense, 2024-04-12:
As a followup to this question, if we want to clone a 3rd party package, I assume we put the git clone command into pre-build.sh.  The question is whether we need to change directory before doing the git clone, so that it ends up in a ros2 workspace.  If so, what is the correct directory?

closed 2024-04-15 by jaybrecht