usnistgov/ARIAC #324: unable to get moveit working in docker

dan9thsense, 2024-04-12:
Trying to set up for the automated evaluation.  I tried installing moveit in the pre-build script with apt install ros-iron-moveit and the package installed but the docker container does not find it for importing.  I tried using source /opt/ros/iron/setup.bash but to no avail.  I also tried installing from source, but the build for that fails.

"ModuleNotFoundError: No module named 'moveit'

jaybrecht, 2024-04-12:
All ros packages will automatically be installed through rosdep if they are added to the `package.xml` 

Try adding these lines to your manifest:

```xml
<depend>moveit_py</depend>
<depend>moveit_core</depend>
<exec_depend>moveit</exec_depend>
```

dan9thsense, 2024-04-12:
That helped but now there was an error in run trial-- it does not launch the sim environment.

from trial_log.txt:
Gazebo Crashed score not recorded

dan9thsense, 2024-04-12:
just for grins, I tried launching it from inside the container:

```
root@db:/workspace/src# source /opt/ros/iron/setup.bash
root@db:/workspace/src# ros2 launch ariac_gazebo ariac.launch.py competitor_pkg:=my_ariac_2024 sensor_config:=barebones trial_name:=kitting dev_mode:=false
Package 'ariac_gazebo' not found: "package 'ariac_gazebo' not found, searching: ['/opt/ros/iron']"
```

dan9thsense, 2024-04-12:
I was able to get it to run inside the container by sourcing both ros iron and the workspace.  I don't know why those are already sourced for nist_competitor but not for when I run my container.

jaybrecht, 2024-04-12:
Couple of things. For the first issues does your launch file follow the steps listed [here](https://pages.nist.gov/ARIAC_docs/en/latest/evaluation/launch_file_setup.html)? The environment launch should be included in your launch file. 

For the second issue the `run_trial.sh` script adds the ROS and workspace source commands in order for everything to run properly. If you open a terminal to the container you will need to add the source commands before launching anything.

dan9thsense, 2024-04-12:
Ah, missed that launch file setup step.  Thanks for the fast response!

closed 2024-04-12 by dan9thsense