usnistgov/ARIAC #325: joints error in docker-- perhaps an old version of competition loaded?

dan9thsense, 2024-04-12:
When running the automated eval, I got an error that I'd seen prior to [pull request 317](https://github.com/usnistgov/ARIAC/pull/317) that fixed a joints issue.  I am not seeing this error when running with the most recent updates.

```
[overall.py-1] [ERROR] [1712952399.955841896] [overall]: in moveit_interface.calculate_joint_error, the number of goal joints = 6
[overall.py-1] [ERROR] [1712952399.957223174] [overall]: is incorrect for the group_name = 'floor_arm', which has 7 joints

```

jaybrecht, 2024-04-12:
Oh yes I need to update the ARIAC image with the new version. I will do that soon. In the meantime you can just do a git pull from the ARIAC folder in the src directory on the container, and rebuild the workspace. That should fix the issue.

dan9thsense, 2024-04-13:
I pulled but it fails to build:

```
root@db:/workspace# colcon build
Starting >>> ariac_msgs
Starting >>> ariac_controllers
Starting >>> ariac_description
Starting >>> ariac_gazebo
Starting >>> ariac_moveit_config
Starting >>> bdai_ros2_wrappers
--- stderr: ariac_description                                                                                                                                                                                            
Traceback (most recent call last):
  File "/opt/ros/iron/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py", line 21, in <module>
    from ament_package.templates import get_environment_hook_template_path
ModuleNotFoundError: No module named 'ament_package'
CMake Error at /opt/ros/iron/share/ament_cmake_core/cmake/ament_cmake_package_templates-extras.cmake:41 (message):
  execute_process(/usr/bin/python3.10
  /opt/ros/iron/share/ament_cmake_core/cmake/package_templates/templates_2_cmake.py
  /workspace/build/ariac_description/ament_cmake_package_templates/templates.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/iron/share/ament_cmake_core/cmake/ament_cmake_coreConfig.cmake:41 (include)
  /opt/ros/iron/share/ament_cmake/cmake/ament_cmake_export_dependencies-extras.cmake:15 (find_package)
  /opt/ros/iron/share/ament_cmake/cmake/ament_cmakeConfig.cmake:41 (include)
  CMakeLists.txt:9 (find_package)

gmake: *** [Makefile:226: cmake_check_build_system] Error 1
---
Failed   <<< ariac_description [0.37s, exited with code 2]
Aborted  <<< ariac_gazebo [0.36s]
Aborted  <<< ariac_moveit_config [0.35s]
Aborted  <<< ariac_controllers [0.39s]
Aborted  <<< bdai_ros2_wrappers [0.67s]                                                            
Aborted  <<< ariac_msgs [1.11s]                            

Summary: 0 packages finished [1.62s]
  1 package failed: ariac_description
  5 packages aborted: ariac_controllers ariac_gazebo ariac_moveit_config ariac_msgs bdai_ros2_wrappers
  5 packages had stderr output: ariac_controllers ariac_description ariac_gazebo ariac_moveit_config ariac_msgs
  4 packages not processed
```

jaybrecht, 2024-04-13:
Source iron first:

`source /opt/ros/iron/setup.bash`

dan9thsense, 2024-04-13:
of course, silly me.

With that it works.

Thanks for the help, especially late on a Friday evening!

closed 2024-04-13 by dan9thsense

jaybrecht, 2024-04-16:
The docker image has been updated.