usnistgov/ARIAC #328: No orders announced after the start of the competition

LKmubihei, 2024-04-15:
First, I start the environment  'ros2 launch ariac_gazebo ariac.launch.py '
Then I start the competition 'ros2 service call /ariac/start_competition std_srvs/srv/Trigger'.but  there is no order is announced in the termial

jfernandez37, 2024-04-15:
Hello. I tried to replicate the issue you are running into, but I could not. Could you tell me a bit more about your setup? What version of ROS are you running? Thanks

jaybrecht, 2024-04-18:
@LKmubihei Were you able to solve this issue?

LKmubihei, 2024-04-19:
There is no order release on my newly installed ubuntu22 system, my ROS2 version is iron, but I can run it fine in docker
[image]

LKmubihei, 2024-04-19:
[image]
After I terminate the program , it displays there is no order announced

LKmubihei, 2024-04-23:
For reasons unknown, when I reinstalled ARIAC again, it could announce orders normally!

jaybrecht, 2024-04-23:
Glad that it worked after a new install. We were unable to replicate the issue so I think it must have been an issue with your local configuration. Let us know if you run into any other issues.

closed 2024-04-23 by jaybrecht