usnistgov/ARIAC #329: playback fails

dan9thsense, 2024-04-16:
Trying to playback a run from automated eval:

```
./playback_trial.sh sirius assembly_3
Successfully copied 77MB to sirius:/home
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
Authorization required, but no authorization protocol specified
[INFO] [1713234091.698937513] [gazebo_ros_node]: ROS was initialized without arguments.
[INFO] [1713234092.060673796] [rclcpp]: signal_handler(signum=2)
```

jaybrecht, 2024-04-16:
Can you try running the command `xhost +local:docker` before the playback? It looks like it is running properly but the gazebo window is just not showing up.

jaybrecht, 2024-04-18:
@dan9thsense Any update on this? We have not been able to replicate this issue on our machine.

dan9thsense, 2024-04-19:
Sorry, been busy with other bugs, just got around to trying it this evening.  Good news, it works. Thanks.