usnistgov/ARIAC #330: attaching a part with two robots causes explosion

dan9thsense, 2024-04-18:
When working on flipped parts and using the illustrated technique of having the two robots do a hand-off of the part, when the part is attached by the second robot, the entire sim immediately explodes most of the time.

dan9thsense, 2024-04-18:
The pump is the worst. The others survive about half the time, but the pump causes the sim to explode every time.

jaybrecht, 2024-04-18:
Are you turning off the gripper of the first robot before turning on the gripper of the other?

Sanchitkedia, 2024-04-19:
When transferring the part from one robot to another, are you updating the planning scene?

dan9thsense, 2024-04-19:
No, I didn't do that.  Why would that help with the explosion?  And when would I do that?  Sometime before the second one attaches?

Sanchitkedia, 2024-04-19:
I tried implementing the method I used last year. It works, but sometimes, the parts don't transfer and slide off. I am not using the "CeilingRobotWaitForAttach" function, hence the slide. I was able to replicate the explosion you are referring to when using the "CeilingRobotWaitForAttach" function. Please feel free to disregard the planning scene part.

dan9thsense, 2024-04-20:
I think the key here is that you deactivate the floor gripper before checking that contact and attachment has been made by the ceiling gripper.  If it is slipping, that means the floor deactivated before the ceiling attached.  In those cases, my code works too.  However, this is not a way that works in the real world.  On the factory floor, safety concerns do not allow losing positive control of the part like that.  Also, the explosion is just a sim artifact.  In the real world, robot arms have enough compliance to allow momentary attachment of two arms simultaneously.

jaybrecht, 2024-04-23:
Yes, I agree Dan. I think this is unfortunately another issue where the gazebo simulation environment has limitations that prevent real world solutions. The way our "vacuum gripper" works is by directly creating a joint between the robot and the part. Also the robots share a robot descriptions so are technically one robot model. Therefore when you try to pick the part using the other robot the robots are fixed together in two places (the world frame and the part frame). I guess the simulation really doesn't like that. 

I know time is limited but it might be worth looking into handling the flip part challenge using a single robot by placing the part on it's side on the AGV and re-picking to flip. This might be more repeatable. 

However, since the issue does not occur every time the fact that trials are run multiple times and we take the best run of the bunch that should prevent the inconsistency from hindering your score too much.

dan9thsense, 2024-04-24:
Good points. I can make it work well enough. Thanks for looking into it.

closed 2024-04-24 by dan9thsense