usnistgov/ARIAC #332: Controlling multiple robots simultaneously

endre90, 2024-04-22:
Hello, I am posting this as it is unclear to me how to advance in order to control multiple robots in Gazebo, i.e. start and stop motion execution whenever I want. I have separated the .srdf files, made different move groups for MoveIt, however, the bottleneck seems to be how the Gazebo environment is set up, namely how to run multiple instances of ros2_control for different robots. I have tried to find answers online, but what I've found is this thread: 
```
https://github.com/usnistgov/ARIAC/issues/210
```
which does provide some example code, but still doesn't solve the underlying issue. Controlling one or the other robot works, but I can't seem to overcome the challenge of controlling the robots in a way where I can issue an request for one robot to move, then do that for the other one while the first one is moving. I've made separate nodes for each robot, and tried with services and actions. The other place I've searched for solutions is at ros2_control, however the only tutorial that is missing is for multiple robot control:
```
https://github.com/ros-controls/ros2_control_demos
```
Any other resources or places to look for a hint? Or is the simulation environment just not meant to be run in such a way?

Best Regards

jaybrecht, 2024-04-23:
So currently simultaneous execution of the robots is not possible using MoveIt directly. If you want to do simultaneous execution you will have to directly send trajectories to the joint_trajectory_controllers for each robot. Since the robot's are in the same planning scene and share a robot description MoveIt is incapable of doing simultaneous execution. You may be able to set it up to still use MoveIt as the planer and then send the trajectories to the controllers directly but that will probably require a modified MoveIt description package.

endre90, 2024-04-23:
Got it, thanks for the feedback!

closed 2024-04-23 by endre90