usnistgov/ARIAC #333: drop part challenge does not drop a part

dan9thsense, 2024-04-23:
I'm trying to test the drop part challenge, but the sim won't drop a part.  The other challenges I've tried work and I used the format shown in the configuration files part of the docs (... drop_after: 0 delay: 0.5) cut and pasted directly, with correct indenting. I can include things in the same trial such as faulty_part and those work without a problem.

Is it working for everyone else?

jaybrecht, 2024-04-23:
We will look into this error. In the meantime can you try using the [trial generator GUI](https://pages.nist.gov/ARIAC_docs/en/latest/competition/configuration_files.html#trial-generator-gui)? This should guarantee that the challenge is output in the correct format in the yaml file.

dan9thsense, 2024-04-24:
Yes, tried that. Part does not drop. Here is the trial generated:

```
# Trial name: dropped_kitting_part.yaml
# ARIAC2024
# Author: Dan
# 2024-04-24 01:17:28

# ENVIRONMENT SETUP

time_limit: -1

# KITTING TRAYS

kitting_trays:
  tray_ids:
  - 1
  slots:
  - 1

# INSERT ROTATION

assembly_inserts:
  as1: 0.0
  as2: 0.0
  as3: 0.0
  as4: 0.0

# PARTS INFORMATION

parts:
  bins:
    bin1:
    - type: pump
      color: blue
      rotation: 0.0
      flipped: false
      slots:
      - 8
    - type: battery
      color: red
      rotation: 0.0
      flipped: false
      slots:
      - 9

# ORDERS INFORMATION

orders:
- id: UYQWVYGB
  type: kitting
  announcement:
    time_condition: 0.0
  priority: false
  kitting_task:
    agv_number: 3
    tray_id: 1
    destination: warehouse
    products:
    - type: battery
      color: red
      quadrant: 1
    - type: pump
      color: blue
      quadrant: 2

# CHALLENGES INFORMATION

challenges:
- dropped_part:
    robot: floor_robot
    type: battery
    color: red
    drop_after: 0
    delay: 0.5
```

jaybrecht, 2024-04-25:
Thanks for spotting this Dan, the linked pull request fixes the issue. We will make sure this is merged and released before qualifiers.

closed 2024-04-25 by jaybrecht