usnistgov/ARIAC #337: clarification on  priority orders

dan9thsense, 2024-04-30:
Does a kitting priority order only need to be submitted before the next kitting order or does any priority order need to be submitted before any other order, regardless of type?

In general, tricky situations arise with the priority orders. Consider the case where a regular order finishes just as a time-based priority order is issued. If those events happen to occur close enough in time, the delays inherent in ROS messages and services mean that the submission can register as occurring after the kitting order is announced and there is no way within the sim environment to anticipate that. 

Perhaps there could be a buffer time, say 10 seconds or so, where submissions can occur after a priority order is announced and still be OK for both orders? That would give enough time to receive the priority order message and stop the submission of a just-completed regular order.

jaybrecht, 2024-04-30:
The rule that priority orders should be submitted before existing orders is not actually enforced by the system or the scoring, it is just in the spirit of the competition to stop work on existing orders when a priority is received. But there are no issues if you submit another order before working on the priority order. 

Due to the priority multiplier it is usually very advantageous to finish the priority order as quickly as possible anyway since it will most likely be worth more points than non-priority orders.

dan9thsense, 2024-04-30:
Got that. But does a kitting priority order only need to be submitted before the next kitting order or does any priority order need to be submitted before any other order, regardless of type, for the priority order to be awarded the priority multiplier?

jaybrecht, 2024-04-30:
No, the priority multiplier is always applied to priority orders regardless of the submission order.

dan9thsense, 2024-04-30:
That's a different answer than than one given in issue #315 where it says the priority multiplier is not given if the priority order is not the next one submitted after being issued.  I've been developing on the basis of that answer.

"The only thing that will happen if the high-priority order is not submitted next is the multiplier being set to 1 instead of 3"

anthonydowns, 2024-05-01:
I think that was a misunderstanding on either your or Zeid’s part.

A priority order will always get the priority multiplier (3) applied to it.

A non-priority order will always get a non-priority multiplier (1) applied to it.

The only difference is how much of an effect this has on your score based on how fast you submit the priority order.

anthonydowns, 2024-05-01:
What Zeid was specifically saying in that answer was that based on your question, if a non-priority order (order 1) is submitted first instead of the priority order (order 2), then the priority factor for order 1 would be set to 1, and the priority factor for order 2 would be set to 3.

This has always been the case for priority orders since the very first year of the competition. Or at the very least, it has always been the intention for the competition to run that way.  The High Priority order is intended to make the competitor system perform an analysis of does it make sense to stop work on this lower priority order and finish the high priority order faster, or vice-versa.

dan9thsense, 2024-05-01:
Thanks for the clarification. Yes, this is what it has always been. The question stemmed from the line in the rules that said the priority order "must" be submitted before other orders. Unfortunately, based on that I thought the rule had changed for this year and I incorrectly interpreted Zeid's answer as saying the priority order would not get that bonus if it was not submitted first.

Of course, having the option to determine when it gets submitted makes for a much more interesting trial.

closed 2024-05-04 by dan9thsense