usnistgov/ARIAC #340: time limit

dan9thsense, 2024-05-04:
Is there any way to know the time limit of an order?

jaybrecht, 2024-05-04:
No, currently there is no way to get the time limit of a trial.

dan9thsense, 2024-05-04:
ok, thanks for the fast response.

dan9thsense, 2024-05-05:
But that means there is no way to know during a robot malfunction or a sensor blackout when to give up and submit the order versus waiting for the challenge to end. While waiting for the challenge to end, the trial time could expire or get so close to expiration that there is not enough time left to complete the trial and it will just expire without even a partial submission. Actually, that's an issue with any challenge that takes a long time to complete.

jaybrecht, 2024-05-05:
I agree and I think it is a feature we would like to add to future versions of the competition. All I would say for now is that the time limit is currently mostly used as an end point to stop the competition when a competitor's system has reached an error and is not able to end the competition properly. We aim to give enough time that all orders can be completed within the time limit of the trial.

dan9thsense, 2024-05-06:
A thought for the future:
Along with the time limit, provide statistics for the frequency and duration of blackouts and malfunctions. Most factories have those numbers. An additional twist would be to pay points for more rapid help. For example, a typical cause of a robot malfunction is that some object fell into the path of the robot. The robot can call for help to get a person to come by and clear the path. In ARIAC, that could be a service call that costs a few points in order to, for example, cut the duration time in half.

Also, right now there is no motivation to end the competition rather than just letting it run to timeout. In fact, there's the cost of doing the code development and a risk that a bug could end the competition early. To encourage competitors to end the competition early, thus saving you time in running all those trials, provide a bonus for ending the competition before the timeout.

closed 2024-05-11 by dan9thsense