usnistgov/ARIAC #343: Add ability to pass in additional build arguments to colcon build step.

AravindaDP, 2024-05-06:
Sometimes teams may need to pass in additional CMake arguments in colcon build step.

e.g. `colcon build --cmake-args -DDOWNLOAD_YOLO_CONFIG=OFF -DENABLE_CUDNN=OFF -DCMAKE_DISABLE_FIND_PACKAGE_OpenMP=TRUE`

This is not supported by currently hard coded colcon build step used by https://github.com/usnistgov/ARIAC_evaluation/blob/main/automated_evaluation/container_scripts/build_competitor_code.py#L81

Current workaround/hack is to use a different git repo (other than what specified in team evaluation configuration file) to host packages with these specific requirements and clone it to somewhere else in the container build it with appropriate flags, and then source it using the pre_build script.

Suggestion
Can it be make possible to include these additional build arguments and append to above colcon build step through team evaluation configuration file?

e.g.

```
# team configuration for automated evaluation

team_name: "test_competitor"

github:
  repository: "github.com/usnistgov/nist_competitor.git"
  tag: "2024.2.2"
  personal_access_token: ""

build:
  pre_build_scripts: ["test_competitor_pre_build.sh"]
  additional_build_args: ["-DENABLE_CUDNN=OFF -DCMAKE_DISABLE_FIND_PACKAGE_OpenMP=TRUE"]

competition:
  package_name: "nist_competitor"
  launch_file: "competition.launch.py"
```

jaybrecht, 2024-05-06:
Yes, we can add this feature. Just upload your yaml file as you have mentioned here with the additional arguments and we will get it to work. We will eventually get it merged into the main ARIAC_evaluation repo so that it can be tested by competitors as well.

AravindaDP, 2024-05-07:
Thanks for the response. For the time being it seems the workaround of pre building with necessary arguments keeps in CMake cache for subsequent builds works.

Having said that one more request is similarly ability to append additional arguments to `rosdep_cmd` So that it's possible to define `--skip-keys` argument to `rosdep`  (Sometimes we have come across situations where we don't want to install dependancies defined in 3rd party ros packages and get away with curated set of packages and then build with additional CMake flags discussed above.) 

If makes sense we can give it a try to jot down a pull request to handle these two features.

jfernandez37, 2024-05-14:
We have added functionality to pass extra arguments into the colcon build command. The changes are in [this pull request](https://github.com/usnistgov/ARIAC_evaluation/pull/9). 

To use this functionality, add an item to the configuration yaml under the `build` section called `extra_colcon_args` with a list of strings for the colcon arguments you would like to pass. You can look at the [nist competitor yaml](https://github.com/usnistgov/ARIAC_evaluation/blob/7f1e02daf5d2b16fda63736083be245196ee5295/automated_evaluation/competitor_configs/nist_competitor.yaml) file for reference. 

Can you please test this with your use case and let us know if you run into any problems? Thanks.

AravindaDP, 2024-05-27:
@jfernandez37 
Apologies for the delayed response. We tested source branch of this PR and covers the requirements we had. Can you confirm whether this would be used in final evaluation setup? If not we could prepare our team configurations based on the alternative workarounds.

We have placed updated configurations in our team folder that can be used to rerun a qualification test.

Team: Runaround Robotics

jfernandez37, 2024-05-28:
Yes. These changes will be used in final evaluation and the pull request was just merged into the main branch.

closed 2024-05-28 by jfernandez37