usnistgov/ARIAC #345: frame_id specified in RGB camera topics do not follow usual convention of a frame aligned with image sensor.

AravindaDP, 2024-05-12:
It seems `frame_id` specified in RGB camera topics (both `rgb_image` and `camera_info` topics) use a frame aligned with camera body rather than optical sensor.

https://pages.nist.gov/ARIAC_docs/en/latest/_images/tf_frames.png

In usual convention, optical frame uses z axis along the forward axis of camera. Here x axis is along the forward axis of cameras.

This causes problems with off the shelf object detection packages.

Either following approach would be OK.

Option 1:
publish rgb/rgbd camera frames oriented in sensor frame orientation. (No change in logical camera frames are expected since they publish the corresponding detection in current frame orientations and they use custom messages)

Option 2:
publish additional frame named `{sensor_name}_optical_frame` as a child frame of camera frame (Currently `{sensor_name}_frame`) with 0 translation and correct orientation transformation. Then use this frame name as `frame_id` in sensor topics.

Approx additional troubleshooting time occurred due to the issue during competition : ~0.5 days

jaybrecht, 2024-05-14:
I can add option 2, a second frame named `{sensor_name}_optical_frame`. Can you specify the exact rotational transform that would need to be applied to the base sensor frame (roll, pitch, yaw)?

AravindaDP, 2024-05-20:
For the time being we adopted using `static_transform_publisher` to get additional tf published ourselves so this is optional but welcome enhancement

For anyone's interested following is the values we used.

```
static_transform_publisher 0 0 0 -1.5708 0 -1.5708 {sensor_name}_frame {sensor_name}_optical_frame
```
i.e. yaw= -pi/2 , pitch = 0, roll = -pi/2

closed 2024-05-28 by jaybrecht