usnistgov/ARIAC #346: OMPL error when picking trays

dan9thsense, 2024-05-14:
The latest pull generates an error when picking trays based on a camera image. Existing code for picking trays is killed by this error.

```
[moveit.ompl_planning.planning_context_manager]: OMPL encountered an error: The fraction of the extent must be larger than 0 and less than 1
```

jaybrecht, 2024-05-14:
I can't think of any changes that have been made that would affect MoveIt or OMPL. Are you using the default moveit_config package or have you made modifications? Looking at this error it seems like it may have to do with the parameter `longest_valid_segment_fraction` as mentioned [here](https://github.com/moveit/moveit/issues/1297). That is defined for each of the planning groups [in this file](https://github.com/usnistgov/ARIAC/blob/ariac2024/ariac_moveit_config/config/ompl_planning.yaml#L78).

dan9thsense, 2024-05-18:
Good call. MoveIt updated at the same time. Reverting to the previous version of MoveIt solved the problem. Sorry to bother you.

closed 2024-05-18 by dan9thsense