usnistgov/ARIAC #348: run_trial continues to crash

dan9thsense, 2024-05-22:
This problem was described in issue #331 which was closed after successful smoke tests (since I didn't close it, I am unable to reopen it, hence this new issue). However, on two separate machines, one with Ubuntu 22.04 newly installed, it continues. Trials that run OK in normal mode, crash gazebo when using docker. However, recently there was one crash that occurred without docker, so it is really hard to know what is going on. The logs are unhelpful, as they just stop at the point of the crash.  

This makes code testing difficult because the run-all option for run_trials.sh fails within one or two trials and, once a trial fails, none of the following succeed. The result is that testing has to be run one trial at a time which means that only a limited number of test trials can be run in a day, as they have to each be hand-loaded.

I understand it does not occur with ARIAC official testing, making it very hard to troubleshoot, but I am wondering if any other teams have encountered it. If not, then I must be making some fundamental mistake in installation. However, I use docker on a regular basis with both machines that have this error and I cannot figure out what is going on here.

Here are a couple typical console outputs:

```
Aborted (core dumped)
Gazebo not running
==== Trial qualifier_trial_3_combined completed
return_code: -9
```

```
[overall.py-20] [INFO] [1716347277.381087254] [overall]: bin 1 has a part-product match, red regulator
[overall.py-20] [INFO] [1716347277.491560870] [overall]: joints move to above_right_part_gripper_changing_station
terminate called after throwing an instance of 'boost::wrapexcept<boost::thread_resource_error>'
  what():  boost thread: trying joining itself: Resource deadlock avoided
Traceback (most recent call last):
  File "/container_scripts/run_trial.py", line 110, in <module>
    main()
  File "/container_scripts/run_trial.py", line 84, in main
    output = subprocess.check_output(
  File "/usr/lib/python3.10/subprocess.py", line 421, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 505, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
  File "/usr/lib/python3.10/subprocess.py", line 1154, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
  File "/usr/lib/python3.10/subprocess.py", line 2022, in _communicate
    self._check_timeout(endtime, orig_timeout, stdout, stderr)
  File "/usr/lib/python3.10/subprocess.py", line 1198, in _check_timeout
    raise TimeoutExpired(
subprocess.TimeoutExpired: Command 'gz topic -l' timed out after 5 seconds

```

dan9thsense, 2024-05-22:
One thought is to just write my own bash script to run all the files and not use docker. However, I cannot figure out how to cleanly shut everything down to move on from one trial to the next.

In a bash script, This fails to kill them all:
    pkill -9 gzclient
    pkill -9 gzserver
    pkill -f "ros2 run"
    pkill -f "ros2 node"
    pkill -f "python3"

Suggestions?

dan9thsense, 2024-05-22:
command line entry to show images from a ros2 camera message

dan9thsense, 2024-05-23:
I found a way to run groups of trials outside docker with a bash script. Since that takes care of the overall issue, I'm closing it now. Nevertheless, it makes me nervous that groups of trials do not effectively run with .run_trials.sh since that is the evaluation method. But it worked in smoke test and quals, so hopefully it will still work in finals.

jaybrecht, 2024-05-23:
Hi Dan, sorry for the late response on this issue. The protobuff error that you mentioned is unfortunately common and we have not been able to determine the cause of this issue. It has existed for a couple years and it seems to be related to gazebo itself. The gazebo messages use protobuff for message transport and something causes a memory overflow in certain situations. From our testing we are able to reduce the occurrence when stopping and restarting the container after each run of the trial. If you take a look at the scoring branch of automated evaluation there is a run_trials.py file. If you run this a GUI will show up that allows you to select the trials you want to perform. This is what we used for the qualifiers. It lets you select the number of completed trials you want to perform and the maximum number of allowed trials. It will rerun a trial if gazebo fails for any reason up to the maximum number of allowed trials. I would try using this tool for extended testing. Let me know if you run into any issues with that.

dan9thsense, 2024-05-25:
I think it helps if there are not too many total parts in the sim. Seems to happen more when the bins have a lot of parts.

dan9thsense, 2024-05-27:
I gave up on getting Docker and run_trials.sh to work consistently. However, I wrote a bash script that works to run batches of trials, so I am able to do testing without Docker.

closed 2024-05-27 by dan9thsense