usnistgov/ARIAC #352: kit_tray tape interfere with Aruco detection?

AravindaDP, 2024-05-24:
Apologies for reporting this late. I thought I had reported this earlier but just now realized I have missed it.

We are using `cv2.aruco.ArucoDetector` with default `cv2.aruco.DetectorParameters` and noticed that sometimes `cv2.aruco.ArucoDetector.detectMarkers(cv_image)` fails to detect tray markers  (cv_image is constructed using `CvBridge.imgmsg_to_cv2(img_msg, desired_encoding="mono8")`)

When we made the tape material less darker by setting ambient and diffuse values as below at https://github.com/usnistgov/ARIAC/blob/ariac2024/ariac_gazebo/models/kit_tray/model.sdf#L39 it started working consistently. (perhaps it could be little bit more darker but we are happy with these values)
```
      <visual name="tape">
        <geometry>
          <mesh>
            <uri>model://kit_tray/meshes/tape.stl</uri>
          </mesh>
        </geometry>
        <material>
          <ambient>0.4 0.4 0.4 1.0</ambient>
          <diffuse>0.4 0.4 0.4 1.0</diffuse>
        </material>
      </visual>
```

I'm not sure it's too late to ask such a change. But appreciate if it could be considered. (Alternatively we would welcome if it can be solved by applying non-default parameters in cv2.aruco detecters as well as a last resort.

Approx additional troubleshooting time occurred due to the issue during competition : ~0.5 days

dan9thsense, 2024-05-25:
Can you explain the change that was made? Our tray detection has been working OK and if you change the sdf now, we have to restart a lot of testing with very little time before finals deadline. What happens if this change makes our model no longer reliably work?

jaybrecht, 2024-05-28:
The color of the tape was change to be gray instead of black. If this causes an issue with your system please let me know and we will figure out a solution. 

[image]

dan9thsense, 2024-05-28:
Was that incorporated into the recent pull request that fixed the flip issue? If so, then we're good, as I used that branch for testing over the weekend and it was ok.

Screenshot.from.2024-05-28.14-47-56.png (view on web)<https://github.com/usnistgov/ARIAC/assets/50928580/2bdfb941-5938-45d1-b62e-3efef9901a8c>

jaybrecht, 2024-05-28:
No, I just merged that PR this morning. The branch 2024.5.0-staging currently has all new changes that will be included in the 5.0 release.

saahu27, 2024-05-29:
The detection using cv2.aruco.ArucoDetector can sometimes fail if you pass in the entire image of the kittray. You can alternatives mask the image with hardcoded pixel coordinates to crop out only the aruco tag and pass that image to opencv aruco detector. This is a better option to detect the Aruco tags.

AravindaDP, 2024-05-30:
Yes, Currently we are testing that approach using the assumption that kit trays are at predesignated coordinates for this year competition. We also noticed that while original color values for kit tray tape material infact wasn't pure black (but still dark enough to confuse ArucoDetector) so color threshold based filtering also should be possible.

So since now deadline is pushed back if other teams have objections to changing kit tray tape color we would be able to adopt one of above alternatives.

PR #353 (2024-05-28): Update kit tray model.sdf
Fixes #352