usnistgov/ARIAC #361: demos?

dan9thsense, 2025-09-17:
There are no demos. The trial opens but no apparent actions.

dan9thsense, 2025-09-17:
Terminal 1:
ros2 launch ariac_gz ariac.launch.py trial_config:={path_to_example_team}/config/trials/LHAF9835.yaml user_config:={path_to_example_team}/config/example_team_config.yaml

Terminal 2:
ros2 run example_team dummy_inspection

Gazebo runs and appears to be OK but the example just hangs saying:
[INFO] [1758127256.879015363] [environment_interface]: Waiting for enviornment to be ready

trivial typo note: "enviornment"  should be "environment"

Same thing with submit_kit, except "environment" is spelled correctly.

trivial typo note: in your reply, "submit kit" should be "submit_kit"

In trying to use the app, it asks for a database and I don't know what to enter. Also, it asks for a package and an executable and has a dropdown list that does not include the new packages-- that's likely a sourcing problem and I will check into that.

jaybrecht, 2025-09-17:
Is the status light in the environment yellow or blue? If it is yellow that means something in the startup did not work properly. 

For the app database you can either use the create button to create an empty db or ignore it. The database is only required to calculate scoring.

dan9thsense, 2025-09-17:
I don't see a status light. Here is a screenshot. Note that the dropdown menus show lots of my ros2 packages and executables but do not include the ones from ariac-- perhaps there is a sourcing issue there.

[image]

When I use the launch file, that is when I get "[INFO] [1758127256.879015363] [environment_interface]: Waiting for enviornment to be ready"

Terminal 1:
ros2 launch ariac_gz ariac.launch.py trial_config:={path_to_example_team}/config/trials/LHAF9835.yaml user_config:={path_to_example_team}/config/example_team_config.yaml

Terminal 2:
ros2 run example_team dummy_inspection

ros2 topic echo /competition_status

competition_state: 0
num_kits: 0
num_modules: 0
num_kits_remaining: 0
num_modules_remaining: 0
time:
  start:
    sec: 0
    nanosec: 0
  elapsed:
    sec: 0
    nanosec: 0
  remaining:
    sec: 100
    nanosec: 0
run_id: 0

dan9thsense, 2025-09-17:
[image]

dan9thsense, 2025-09-17:
Ah, perhaps you meant the color of the "prepare" label. Gazebo launches but the status light hangs on yellow and the message competition_status/competition state remains 0

dan9thsense, 2025-09-17:
Solved the source problem but it still hangs.

[image]

jaybrecht, 2025-09-17:
> Solved the source problem but it still hangs.
> 
> <img alt="Image" width="1091" height="724" src="https://private-user-images.githubusercontent.com/1042607/490690570-c3739657-f111-41bc-8e89-af6253a1b0fe.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTgxMzE0MjMsIm5iZiI6MTc1ODEzMTEyMywicGF0aCI6Ii8xMDQyNjA3LzQ5MDY5MDU3MC1jMzczOTY1Ny1mMTExLTQxYmMtOGU4OS1hZjYyNTNhMWIwZmUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDkxNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA5MTdUMTc0NTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2E2MWUxOTc5MzVhNTE2ODc0YjFjMjQ2NDlhMzA4YzM2MzFmZDk1ZWI1YTIzODRkMzZjNDJhMzhjN2YwNDA4ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Zef5lipWxvAh9kQZQgbS3LCl8qq9tu7B_fSNzmhkuY8">

For the ARIAC app I can see how the layout is a bit confusing. The green start run button starts Gazebo with the settings configured on the home page. Then in addition to that you can start a secondary ros process which in this case is the example team demo by pressing the small arrow button next to the command. And yes as you figured out you have to source the example team package before launching the app in order for those to show up.

dan9thsense, 2025-09-17:
Ah, OK. Here is the status light. I'll try that and look for a missing dependency. 

[image]

jaybrecht, 2025-09-17:
Improved instructions for [startup](https://pages.nist.gov/ARIAC_docs/en/latest/pages/startup.html) and instructions for the [demos](https://pages.nist.gov/ARIAC_docs/en/latest/pages/demos.html) is now available.