usnistgov/ARIAC #363: defective cells defined?

dan9thsense, 2025-09-20:
Are there criteria defined for what is a defective cell that the lidar should detect? I recall a slide from the introduction meeting showing what I think was a dent but kind of looked like a smudge.

dan9thsense, 2025-09-24:
Thanks for providing details. Can the defect be on the top of the cell or only along the cylindrical wall? If only along the wall, what is the purpose of the "C" area for lidars?

dan9thsense, 2025-09-24:
oh, and can  lidars "interfere" with each other if they are pointed at each other? That happens with a real lidar if it picks up stray beams from another unit.

jaybrecht, 2025-09-24:
Defects will only occur on the cylindrical surface of the shell, not on the top of the cell.

The C bounding box is provided for teams that want to position sensors in line with the conveyor. However, these sensors will need to be angled downward to function properly. Since sensors don't have collision geometry, we wanted to prevent situations where teams might place sensors directly in the conveyor path where cells would pass through them.

The lidar sensors operate independently and will not interfere with each other electronically. However, there is a potential issue to be aware of: if lidar sensors are positioned where they're in range of each other, their physical geometry can appear in the point cloud data. This applies to all ray-based sensors (break-beam, distance, and lidar) since they operate based on visual geometry rather than collisions.